mne-tools/mne-python #2805: BUG: pick_types with KIT

kingjr, 2016-01-20:
In `pick_type()` when there's a bad channel:

```
from mne.channels import find_layout
from mne.io import read_raw_kit
fname = 'mne-python/mne/io/kit/tests/data/test.sqd'
raw = read_raw_kit(fname, preload=True)
print('raw', find_layout(raw.info))
raw.pick_types(meg=True)
print('raw pick_type', find_layout(raw.info))
raw.info['bads'] += ['MEG 002']
print('raw bad', find_layout(raw.info))
raw.pick_types(meg=True)
print('raw bad pick_type', find_layout(raw.info))
```

gives:

```
('raw', <Layout | KIT-157 - Channels: MEG 001, MEG 002, MEG 003 ...>)
('raw pick_type', <Layout | KIT-157 - Channels: MEG 001, MEG 002, MEG 003 ...>)
('raw bad', <Layout | KIT-157 - Channels: MEG 001, MEG 002, MEG 003 ...>)
('raw bad pick_type', None)
```

This makes the whole plotting function incompatible once a sensor is rejected.

@Eric89GXL @teonlamont

teonlamont, 2016-01-20:
Yeah, I came across this problem too...temp workaround is to interpolate bad channels then the layout will work. The problem I had when working on a patch is dealing with the sparse matrix operations like removing a channel are not easy

agramfort, 2016-01-21:
the find_layout function should be robust to a small fraction of bad channels

kingjr, 2016-01-21:
I didn't to fix it. @teonlamont are you up for it?

teonlamont, 2016-01-22:
@kingjr unfortunately I can't. I'm swamped :( can't really make contributions (just more issues) until after D-Day :sweat_smile:

jaeilepp, 2016-01-22:
I could make a simple fix. You'll have to see what you think about it...

jaeilepp, 2016-01-25:
Fixed.

closed 2016-01-25 by jaeilepp