mne-tools/mne-python #2810: BUG: make_forward_solution n_jobs=-1

kingjr, 2016-01-21:
When running `make_forward_solution` with `n_jobs=-1`, 

```
/home/jrking/mne-python/mne/forward/_make_forward.pyc in make_forward_solution(info, trans, src, bem, fname, meg, eeg, mindist, ignore_ref, overwrite, n_jobs, verbose)

/home/jrking/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    549         finally:
    550             set_log_level(old_level)
--> 551     return function(*args, **kwargs)
    552 
    553 

/home/jrking/mne-python/mne/forward/_make_forward.pyc in make_forward_solution(info, trans, src, bem, fname, meg, eeg, mindist, ignore_ref, overwrite, n_jobs, verbose)
    553         update_kwargs, bem = _prepare_for_forward(
    554             src, mri_head_t, info, bem, mindist, n_jobs, bem_extra, trans,
--> 555             info_extra, meg, eeg, ignore_ref, fname, overwrite)
    556     del (src, mri_head_t, trans, info_extra, bem_extra, mindist,
    557          meg, eeg, ignore_ref)

/home/jrking/mne-python/mne/forward/_make_forward.pyc in _prepare_for_forward(src, mri_head_t, info, bem, mindist, n_jobs, bem_extra, trans, info_extra, meg, eeg, ignore_ref, fname, overwrite, verbose)

/home/jrking/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    549         finally:
    550             set_log_level(old_level)
--> 551     return function(*args, **kwargs)
    552 
    553 

/home/jrking/mne-python/mne/forward/_make_forward.pyc in _prepare_for_forward(src, mri_head_t, info, bem, mindist, n_jobs, bem_extra, trans, info_extra, meg, eeg, ignore_ref, fname, overwrite, verbose)
    438     if not bem['is_sphere']:
    439         inner_skull = _bem_find_surface(bem, 'inner_skull')
--> 440         _filter_source_spaces(inner_skull, mindist, mri_head_t, src, n_jobs)
    441         logger.info('')
    442 

/home/jrking/mne-python/mne/source_space.pyc in _filter_source_spaces(surf, limit, mri_head_t, src, n_jobs, verbose)

/home/jrking/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    549         finally:
    550             set_log_level(old_level)
--> 551     return function(*args, **kwargs)
    552 
    553 

/home/jrking/mne-python/mne/source_space.pyc in _filter_source_spaces(surf, limit, mri_head_t, src, n_jobs, verbose)
   1998 
   1999         # Check that the source is inside surface (often the inner skull)
-> 2000         outside = _points_outside_surface(r1s, surf, n_jobs)
   2001         omit_outside = np.sum(outside)
   2002 

/home/jrking/mne-python/mne/source_space.pyc in _points_outside_surface(rr, surf, n_jobs, verbose)

/home/jrking/mne-python/mne/utils.pyc in verbose(function, *args, **kwargs)
    549         finally:
    550             set_log_level(old_level)
--> 551     return function(*args, **kwargs)
    552 
    553 

/home/jrking/mne-python/mne/source_space.pyc in _points_outside_surface(rr, surf, n_jobs, verbose)
   2047     parallel, p_fun, _ = parallel_func(_get_solids, n_jobs)
   2048     tot_angles = parallel(p_fun(surf['rr'][tris], rr)
-> 2049                           for tris in np.array_split(surf['tris'], n_jobs))
   2050     return np.abs(np.sum(tot_angles, axis=0) / (2 * np.pi) - 1.0) > 1e-5
   2051 

/home/jrking/anaconda/lib/python2.7/site-packages/numpy/lib/shape_base.pyc in array_split(ary, indices_or_sections, axis)
    408         Nsections = int(indices_or_sections)
    409         if Nsections <= 0:
--> 410             raise ValueError('number sections must be larger than 0.')
    411         Neach_section, extras = divmod(Ntotal, Nsections)
    412         section_sizes = ([0] +

ValueError: number sections must be larger than 0.
```

When setting the `n_jobs` to 2 it works.

Eric89GXL, 2016-01-21:
I actually hit this and fixed it in my movecomp PR, should land in master
soon

kingjr, 2016-01-21:
Ok thanks

closed 2016-01-21 by kingjr