mne-tools/mne-python #2814: EHN: explicit test for Freesurfer & MNE environment variables?

kingjr, 2016-01-22:
I regularly encounter people who run into highly non explicit `subprocess.call()` errors due to inexistent environment variable and source (`FREESURFER_HOME`, 'MNE_ROOT`) or missing Freesurfer licence.

Today, the error was due to the fact that Spyder didn't have the same env variable as the terminal. 

In any case, I think it'd be very helpful if we had a little function that checks i) whether the necessary env variables are present and ii) whether they point to a real path, and throw an error otherwise.

agramfort, 2016-01-22:
can you give a function that does not check this early enough?

Eric89GXL, 2016-01-22:
We already have something like this for `subjects_dir` I think, so we can generalize that function for this purpose. Although in the case of MNE and FreeSurfer, we could verify paths a bit better e.g. by looking for files we know should be close to the root of the install, so maybe separate functions would be best.

kingjr, 2016-01-22:
The case I encountered today was `make_watershed_bem` IIRC. (Not my computer though, so you won't have much more details.)

In sort of the same topic, I personally check that the subject has been correctly segmented with from the log of freesurfer recon-all. But this may be too stringent depending on how users ran their segmentations.

```
def check_freesurfer(subjects_dir, subject):
    # Check freesurfer finished without any errors
    fname = op.join(subjects_dir, subject, 'scripts', 'recon-all.log')
    if op.isfile(fname):
        with open(fname, 'rb') as fh:
            fh.seek(-1024, 2)
            last = fh.readlines()[-1].decode()
        print last
        print('{}: ok'.format(subject))
        return True
    else:
        print('{}: missing'.format(subject))
        return False
```

agramfort, 2016-01-22:
what you report weird:

https://github.com/mne-tools/mne-python/blob/master/mne/bem.py#L956

no?

kingjr, 2016-01-22:
Unfortunately, I can't tell you much more. The error was in Spyder which I don't have. Also, adding env vars in Spyder via `mne.utils.set_config()` didn't work, but `os.env['']` did. I'm not sure how spyder deals with env variables.

Two additional errors followed:
- missing licence file despite `export FREESURFER=path/to/freesurfer/` 
- no writing permission in `path/to/freesurfer/subjects` after freesurfer fresh install on a mac.

Both errors crashed non-explicitely after subprocess.call()

Eric89GXL, 2016-01-22:
> Also, adding env vars in Spyder via mne.utils.set_config() didn't work

Just to be clear, `mne.utils.set_config(...)` does not set environment variables, it sets variables in the mne-python config file. So setting those will only have an effect when mne-python goes to query variables, not when external programs do.

I guess we could have mne-python query these vars and pass them, if available, in the `run_subprocess` call if that would make things easier.

> I'm not sure how spyder deals with env variables.

Spyder deals with env variables the same way that any program does, which depends on the OS and how it is launched. If you launch it from a terminal, it should inherit the env of the terminal. If you launch it from the OS's GUI, then it will probably only have the global vars (i.e., not those from any particular user's .bashrc or .cshrc file).

Eric89GXL, 2016-01-22:
> * missing licence file despite export FREESURFER=path/to/freesurfer/
> * no writing permission in path/to/freesurfer/subjects after freesurfer fresh install on a mac.

I suspect these are both issues with the FREESURFER install rather than MNE. Can the user use FreeSurfer properly outside of the mne-python context?

kingjr, 2016-01-22:
> I suspect these are both issues with the FREESURFER install rather than MNE. Can the user use FreeSurfer properly outside of the mne-python context?

Absolutely, these are Freesurfer issues; but it was difficult to track down for the user, because the thrown error was cryptic.

Eric89GXL, 2016-02-03:
FYI @kingjr it's "ENH" not "EHN", looks like a few of your issues have this issue :)

kingjr, 2016-02-03:
yep, definitely a systematic error, my bad

jasmainak, 2016-02-04:
I remember it took me a while to figure out what ENH stood for ;)

kingjr, 2016-06-01:
> can you give a function that does not check this early enough?

`mri_watershed` called by `mne.bem.make_watershed_bem` (identified in https://github.com/mne-tools/mne-python/issues/3274)

agramfort, 2016-06-01:
the _prepare_env in bem.py is meant to check early. it's not good enough?
if not PR welcome

kingjr, 2016-06-01:
> the _prepare_env in bem.py is meant to check early. it's not good enough?

Apparently not.

I started a PR to try to automatically detect freesurfer directory in case the user hasn't given the right env variables (or whatever other reason, such as conflict between profile sources).

I'm nearly there:
https://gist.github.com/kingjr/97d19a1d5d2af7b2f7e5b9b7badfe4d2

But didn't manage to capture the failure to launch `SetUpFreeSurfer.sh`, and I'm spending way too much time on this, so I'll just leave it there until we find the motivation to finish it.

larsoner, 2018-04-04:
I think we have better checks now, feel free to reopen if they are still insufficient

closed 2018-04-04 by larsoner