mne-tools/mne-python #2816: FIX: report too many sections hides page

kingjr, 2016-01-22:
@mainakjas @dengemann 

When there's a too many section button for one line, the button start to hide the top of the page.

here's an extreme example

[image]

jasmainak, 2016-01-22:
I suspect this line needs to be fixed: https://github.com/mne-tools/mne-python/blob/master/mne/report.py#L533. Basically change it to percentage. @kingjr do you want to give it a try? You can reproduce the problem better than me.

kingjr, 2016-01-22:
ok

kingjr, 2016-01-22:
... actually no; I don't know. The margin should change according to the number of line used;

jasmainak, 2016-01-22:
can you give me a gist to reproduce? I'll fix it then.

jasmainak, 2016-01-22:
By the way, do you notice the `>` on the top left? You can use it to hide the section buttons. I think it was meant to fix this exact issue. That doesn't do it for you?

kingjr, 2016-01-23:
> By the way, do you notice the > on the top left? You can use it to hide the section buttons. I think it was meant to fix this exact issue. That doesn't do it for you?

Aaah! nice. It's good enough, I'll close it :)

closed 2016-01-23 by kingjr