mne-tools/mne-python #2825: unexpected data array size after crop

barry, 2016-01-27:
Hi,

I am relatively new to mne-python, so I apologize if this has an obvious explanation. The array shape of a crop of a raw data file seems to depend on the time-point that is cropped. 

For example:

    from __future__ import print_function
    import mne
    import os.path as op
    
    data_path = op.join(mne.datasets.sample.data_path(), 'MEG',
    'sample', 'sample_audvis_raw.fif')
    raw = mne.io.RawFIF(data_path, preload=True, verbose=False)
    tmins = [40*x for x in range(7)]
    for tmin in tmins:
        raw1 = raw.crop(tmin=tmin, tmax=tmin + 2, copy=True)
        print(raw1._data.shape)

returns: 

    (376, 1202)
    (376, 1202)
    (376, 1202)
    (376, 1202)
    (376, 1203)
    (376, 1204)
    (376, 1204)

Even allowing for rounding, I would not expect raw1._data.shape[1] to vary by more than 1 (and in longer fif files the the size of _data continues to grow linearly as a function of tmin).

Many thanks,

Barry.

agramfort, 2016-01-27:
the rounding is done based on nearest sample for both tmin and tmax. you
can there have a +-1 for tmin and for tmax.

there's no bug.

close the issue if you agree.

Eric89GXL, 2016-01-27:
> the rounding is done based on nearest sample for both tmin and tmax.

FYI this is true if you are on `master` -- for the latest release (0.11) it was based on absolute bounds (up to numerical precision).

barry, 2016-01-28:
A +/- 1 discrepancy due to rounding makes sense, but as I mentioned in my first post the discrepancy scales linearly with tmin and so the discrepancy can be arbitrarily large.

Here is what I see with a longer recording:

```
raw = mne.io.RawFIF(fif_path, preload=True, verbose=False)
print(raw.info['sfreq'])
print(raw._data.shape)
tmins = [40*x for x in range(14)]
for tmin in tmins:
    raw1 = raw.crop(tmin=tmin, tmax=tmin + 2, copy=True)
    print(raw1._data.shape)
```

1000.0
(401, 820000)
(401, 2001)
(401, 2001)
(401, 2001)
(401, 2003)
(401, 2003)
(401, 2005)
(401, 2005)
(401, 2005)
(401, 2007)
(401, 2007)
(401, 2009)
(401, 2009)
(401, 2009)
(401, 2011)

`>>> mne.__version__`
'0.11.dev0'

agramfort, 2016-01-28:
that's weird and looks indeed like a bug.

can you give a try to bug hunting by looking at

https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L1100

?

thx

jaeilepp, 2016-01-28:
I think I can fix this...

agramfort, 2016-01-28:
go for it !

closed 2016-02-02 by Eric89GXL