mne-tools/mne-python #2828: add option in "mne report" to get sensitivity maps of forward models

agramfort, 2016-01-30:
it's a good quality insurance plot for forward models.

agramfort, 2016-01-30:
something along these lines (copied from a script of @dengemann)

```
mag_map = mne.sensitivity_map(fwd, ch_type=fwd_ch_type, mode=fwd_ori)
brain = mag_map.plot(subject=subject, time_label='Magnetometer sensitivity',
                     hemi='rh', subjects_dir=subjects_dir, **fwd_plot_args)
brain.save_image('sensitivity_mag.png')
report.add_images_to_section('sensitivity_mag.png', 'sensitivity map',
                             'forward')
os.remove('sensitivity_mag.png')

fig = plt.figure()
plt.hist(mag_map.data.ravel(),
         bins=20, label=['Magnetometers'],
         color=['b'])
plt.title('Normal orientation sensitivity')
plt.xlabel('sensitivity')
plt.ylabel('count')
plt.legend()
```

Eric89GXL, 2016-09-08:
@jasmainak do you have time for this in the next two weeks?

jasmainak, 2016-09-09:
okay I'll give some time for fixing MNE stuff on Monday. I've a couple of things to do.

Eric89GXL, 2016-09-16:
@jasmainak any time for this one in the next couple of days? If not I'll remove the milestone (or you can)

jasmainak, 2016-09-16:
I have a local branch @Eric89GXL but I fear it's going to take more time. I'll remove it from the milestone

parulsethi, 2017-03-05:
hi,
I’m interested in solving this issue, but will need some help through it.

If I understand correctly, a Report class method needs to be added to append 'sensitivity map' to the report? and is ‘sensitivity/count histogram’ mentioned above also required to be appended?

and if a method, will the input parameters to it would be same as that of 'mne.sensitivity_map()+SourceEstimate.plot()' or something else?

jasmainak, 2017-03-05:
hi @parulsethi I'd suggest that you start from my work in progress commit [here](https://github.com/jasmainak/mne-python/commit/2405a89c740b041758622fdf61e00adf71ba098b). I don't think I got it working, but it might be easier than starting from scratch.

jasmainak, 2017-03-05:
Also, make sure you understand how to generate a report. You can read the tutorial on the mne report [here](http://martinos.org/mne/dev/tutorials/report.html)

parulsethi, 2017-03-05:
@jasmainak thanks for the suggestion, I've forked from there.

larsoner, 2018-04-04:
@agramfort do we still need this? With the ease of adding sections manually it does not seem essential

agramfort, 2018-04-04:
I would say so as it's the visual feedback on quality assessment of forward
computation.

sappelhoff, 2021-03-19:
@drammock is this really in progress? It seems to me that @jasmainak self-assigned in 2016 :-) I think that's no longer up to date.

I think this is still "up for grabs"

drammock, 2021-03-19:
> @drammock is this really in progress? It seems to me that @jasmainak self-assigned in 2016 :-) I think that's no longer up to date.
> 
> I think this is still "up for grabs"

I was just hoping to inspire @jasmainak to finally get to it 😆

jasmainak, 2021-03-19:
I did say, I'll get to it by Monday .......

2021? :)

1himan, 2026-02-27:
would a PR still be welcomed on this issue? @drammock