mne-tools/mne-python #2829: EHN: support MRI viewer

kingjr, 2016-01-30:
I need an MRI viewer to detect (or rather infer) the locations of intracranial electrodes and add some fitting routing to help the user.

I think the MNE C has this functionality (for the coregistration) but this is not the case in MNE-Python, right?

I tried to see whether it'd be possible to use nilearn plotting functions, but [they are not really up for interactive plotting](https://github.com/nilearn/nilearn/issues/969) and redirected me to a [useful, although pretty dead rep: `pynax`](https://github.com/AlexandreAbraham/pynax)

I think the MRI viewer will be useful for several use cases, so it might be worth adding such thing in the future. The TODO list to complete pynax would be:
* plot in mm space instead of voxel space (using `nilearn.affine`?)
* have buttons to adjust contrast, luminosity and zoom
* have a 4th panel (on top of x, y z MRI slices) showing the BEM in 3D with the 3 planes of 2D view.
In parallel, I'll try to work on specifically on methods to help electrode locations. This is likely to push intracranial people to MNE, because the current methods based on MR (and not CT) are just aweful (I'm just coming back from a tutorial based on a combination of Excel sheet, mricron, FSL and Freesurfer simultaneously opened).

Do you think we should try to work on this in MNE, or have a separate rep?

@choldgraf

choldgraf, 2016-01-30:
How common is it to use only MR-based methods? I don't think I've ever seen an ECoG case that had an MR but no CT. Either way, this would be useful for any intracranial people. It'd be useful to include some electrode selection tool as well (as a first pass maybe you can supply a list of channel names and it will let you specify the location for each one?)

kingjr, 2016-01-30:
They don't have any CT at NYU (one of the largest intracranial center), and neither did they in Paris (pretty small intracranial center).

Also, the good thing with MR location is that you can check online that the electrodes are on the surface of the cortex.

>  It'd be useful to include some electrode selection tool as well (as a first pass maybe you can supply a list of channel names and it will let you specify the location for each one?)

Absolutely, I was thinking of having a button that increment a list of electrode arrays. You would add an array by specifying the number and arrangements of electrodes (e.g. 1 strip of 8 electrodes interspaced by 10 mm, or a grid of 4 * 8)  and indicate whether it is rigid (depth electrode) or flexible (surface) so has to fit a line of a spline to automatically propose all the location of the array from 2 or 3 inputs.

choldgraf, 2016-01-30:
Interesting - I didn't realize that they don't use CT scans. What do you mean you can check "online"? As in, intraoperatively?

Re: electrode selection, that sounds good to me. Could be worth implementing a `Grid` and `Depth` class, if anything just to be containers for the proper information for each set of channels. I think it could be useful for handling the viz / surface snapping stuff, though maybe is overkill for the MNE package more generally.

agramfort, 2016-01-30:
I'd recommend to start this in a separate repo and maybe build on top of
pynax.

A

Eric89GXL, 2016-02-01:
Also `nibabel` almost had some level of visualization support for viewing slices:

https://github.com/nipy/nibabel/pull/251

I wonder if we should start our own separate package for this sort of thing.

dgwakeman, 2016-02-01:
This would be a good project to coordinate with @aestrivex and Noam Peled. @aestrivex has built a tool with visualizes and even autodetects intracranial electrodes (there is a manuscript in  prep).

kingjr, 2016-02-12:
Thanks @Eric89GXL for the link. I eventually used your viewer, as it was cleaner. (The main difficulty is to go back and forth from head coordinates to image coordinates.)

I've started a separate rep here https://github.com/kingjr/ecoggui as this seemed a bit too disjoint from MNE. It is still preliminary, but it works (on my computer..).

The main blocker is that my electrode fitting is way too slow, but I implemented my own loss function, so I expect that it improvable by several orders of magnitudes... (if there is any optimization experts in the room... ... ... too busy? ok.)

@choldgraf Let me know whether you're still up for helping ;).

closed 2016-02-12 by kingjr

choldgraf, 2016-02-13:
@kingjr my workload should lighten up in a few weeks or so...keep me in the loop and I will help out once I've got the time!

kingjr, 2016-02-13:
Sure. Here's an animated gif to persuade you to join in earlier than later.

[image]

Isn't that persuasive? :]

choldgraf, 2016-02-13:
hehe - is that one of the new matplotlib colormaps I see? That is
compelling indeed :)

Also, this is the first I've seen someone selecting electrode positions
based on the MRI dropout. Cool!

kingjr, 2016-02-15:
maybe we should write a paper then ;)

choldgraf, 2016-02-16:
Actually for a long time I've wanted to write a "best practices in ECoG"
kind of paper (or set of blog posts, or something)...right now it seems
like everybody just goes with whatever works best for them :P