mne-tools/mne-python #2831: FIX: some examples seem broken

dengemann, 2016-02-01:
but I cannot reproduce it locally

http://martinos.org/mne/dev/auto_examples/plot_compute_mne_inverse.html#sphx-glr-auto-examples-plot-compute-mne-inverse-py

http://martinos.org/mne/stable/auto_examples/inverse/plot_label_source_activations.html#sphx-glr-auto-examples-inverse-plot-label-source-activations-py

http://martinos.org/mne/dev/auto_examples/visualization/plot_meg_eeg_fields_3d.html#sphx-glr-auto-examples-visualization-plot-meg-eeg-fields-3d-py

http://martinos.org/mne/dev/auto_examples/inverse/plot_make_inverse_operator.html#sphx-glr-auto-examples-inverse-plot-make-inverse-operator-py

Eric89GXL, 2016-02-01:
Ouch! @jaeilepp did you make these or did I?

jaeilepp, 2016-02-02:
I had some weird testing dataset a while ago, so these might be mine. If we get https://github.com/mne-tools/mne-python/pull/2761 working, we could get circleci up and running soon.

jaeilepp, 2016-02-02:
And I think we'll have to look at some of the examples. They take awfully long time to run.

Eric89GXL, 2016-02-03:
Closing as dup of #2757

closed 2016-02-03 by Eric89GXL