mne-tools/mne-python #2834: CSP Test Fail

clamus, 2016-02-02:
Hi Guys, The CSP test is failing on my machine. I am using the dev version of mne-python, and have not modified anything for this test. I am running OS X with anaconda with mkl. Any suggestions/ideas as to why this is happening? Below is the message I get.
Thanks!
======================================================================
FAIL: Test Common Spatial Patterns algorithm on epochs
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/lamexicana/anaconda/lib/python2.7/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/Users/lamexicana/repos/mne-python/mne/decoding/tests/test_csp.py", line 76, in test_csp
    assert_array_almost_equal(csp.filters_, csp_epochs.filters_, -1)
  File "/Users/lamexicana/anaconda/lib/python2.7/site-packages/numpy/testing/utils.py", line 842, in assert_array_almost_equal
    precision=decimal)
  File "/Users/lamexicana/anaconda/lib/python2.7/site-packages/numpy/testing/utils.py", line 665, in assert_array_compare
    raise AssertionError(msg)
AssertionError: 
Arrays are not almost equal to -1 decimals

(mismatch 11.1111111111%)
 x: [repr failed]
 y: [repr failed]
-------------------- >> begin captured stdout << ---------------------
Opening raw data file /Users/lamexicana/repos/mne-python/mne/io/tests/data/test_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
Current compensation grade : 0
    Range : 25800 ... 40199 =     42.956 ...    66.930 secs
Ready.
Adding average EEG reference projection.
1 projection items deactivated
15 matching events found
Created an SSP operator (subspace dimension = 3)
4 projection items activated
Reading 2425 ... 2845  =      4.038 ...     4.737 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 2851 ... 3271  =      4.747 ...     5.446 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 4105 ... 4525  =      6.835 ...     7.534 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 4530 ... 4950  =      7.542 ...     8.242 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 5745 ... 6165  =      9.565 ...    10.264 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 6181 ... 6601  =     10.291 ...    10.990 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 7405 ... 7825  =     12.329 ...    13.028 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 7792 ... 8212  =     12.973 ...    13.673 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 9036 ... 9456  =     15.045 ...    15.744 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 9508 ... 9928  =     15.830 ...    16.530 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 10656 ... 11076  =     17.742 ...    18.441 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 11087 ... 11507  =     18.459 ...    19.159 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 12406 ... 12826  =     20.655 ...    21.355 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 12791 ... 13211  =     21.297 ...    21.996 secs...
[done]
Applying baseline correction ... (mode: mean)
Reading 14006 ... 14399  =     23.319 ...    23.974 secs...
[done]
Applying baseline correction ... (mode: mean)
1 bad epochs dropped

--------------------- >> end captured stdout << ----------------------
>>  raise AssertionError('\nArrays are not almost equal to -1 decimals\n\n(mismatch 11.1111111111%)\n x: [repr failed]\n y: [repr failed]')
    

----------------------------------------------------------------------
Ran 50 tests in 49.692s

FAILED (SKIP=4, failures=1)

jasmainak, 2016-02-02:
Can you do `git log` and share what you see?

clamus, 2016-02-02:
Here is the first few line:
(conda)MacBook-Pro:mne-python lamexicana$ git log
commit 5b00a72cbd43a699d79d79ee0334d5973682b2f8
Merge: 2c802db f05a55a
Author: Alexandre Gramfort <alexandre.gramfort@m4x.org>
Date:   Mon Feb 1 19:01:33 2016 +0100

    Merge pull request #2830 from Eric89GXL/noproj
    
    MRG: Fix projector removal

commit f05a55aaf45d2625d0f54fb55ee5352f49c46f76
Author: Eric Larson <larson.eric.d@gmail.com>
Date:   Mon Feb 1 12:07:04 2016 -0500

    FIX: Fix projector removal

commit 2c802db1ca85a76320dad4c6b97e4ea3443747fc
Merge: 9cb2846 fe1ae92
Author: Alexandre Gramfort <alexandre.gramfort@m4x.org>
Date:   Sat Jan 30 09:20:25 2016 +0100

    Merge pull request #2758 from jasmainak/speedup_interp
    
    MRG: spline interp fixes

commit fe1ae92e1be8df61f96007a445ddd3e72bdfda79
Author: Mainak Jas <mainak.jas@telecom-paristech.fr>
Date:   Fri Jan 29 20:31:51 2016 +0100

    Address comments + fixes

commit 14ebd0ee7550c48043e2e03407a1475021e434df
Author: Mainak Jas <mainak.jas@telecom-paristech.fr>
Date:   Wed Jan 27 14:35:51 2016 +0100

    More fixes

commit 5e7c964a45e2289f464c2435028c6b21661c301e
Author: Mainak Jas <mainak.jas@telecom-paristech.fr>
Date:   Wed Jan 27 14:52:07 2016 +0100

    FIX: make interp code match 1989 paper

commit 0cc42b495bbb6b73e8f0e4fdbcad52f6d1d74ae8
Author: Mainak Jas <mainakjas@gmail.com>
Date:   Fri Jan 1 20:31:44 2016 +0530

    MAINT: name the file channel_interpolation

commit b766da987cde446cd9e9ef6e4175abdc69ba85f4
Author: Mainak Jas <mainakjas@gmail.com>
Date:   Thu Dec 31 18:52:24 2015 +0530

    DOC: Add explanation of interpolation code

commit 3cfb8a48c1c168161f43c3d9c304144b3718964f
Author: Mainak Jas <mainakjas@gmail.com>
Date:   Thu Dec 31 16:47:03 2015 +0530

    MAINT: remove superfluous computation

commit 9cb28460b028eabccc305ae80c7f32ca88ba5b68
Merge: 417878d fa0afb2
Author: Alexandre Gramfort <alexandre.gramfort@m4x.org>
Date:   Thu Jan 28 11:16:47 2016 +0100

    Merge pull request #2824 from jona-sassenhagen/patch-8
    
    DOC: evoked.plot_ plot lim docs

commit fa0afb23387973efeacab8ffc2048aba955efd26
Author: jona-sassenhagen <jona.sassenhagen@gmail.com>
Date:   Wed Jan 27 17:58:15 2016 +0100

    evoked.plot_ plot lim docs
    
    Fix docstring for evoked plotting lims. Alternative to #2823
    
    (Sorry if this isn't the proper way ...)
    
    @jasmainak @teonlamont

commit 417878d0d04ac88e78d3cc69ef06465c0ed3b4fa
Author: Eric Larson <larson.eric.d@gmail.com>
Date:   Fri Dec 18 11:29:15 2015 -0500

    FIX: Add montage reading

commit 3786fc818409e27d01e78b6e1e575c01548649e6
Author: Eric Larson <larson.eric.d@gmail.com>
Date:   Wed Dec 16 13:53:21 2015 -0500

    ENH: Add FIF montage reading

jasmainak, 2016-02-03:
Indeed, I can confirm that this is a problem

clamus, 2016-02-03:
Thanks for the confirmation. Hum, do you is it related to osx+anaconda+mkl?

jasmainak, 2016-02-03:
I am on Linux Mint and can reproduce on this. We need to trace the commit that broke it.

jasmainak, 2016-02-03:
Looks like the test was always broken. This is the PR: https://github.com/mne-tools/mne-python/pull/2666 and the related commit: https://github.com/mne-tools/mne-python/commit/32aa9f8ef5059e170f1cad3443a2b01279655496. @agramfort and @cle1109 do we need to change the tolerance?

jasmainak, 2016-02-03:
@Eric89GXL why is the `test_csp` classified as a slow test? It seems to run pretty fast for me. We didn't detect the issue for 3 months ...

Eric89GXL, 2016-02-03:
`slow` tests still get run by `nosetests` and most of the Travis builds (but not AppVeyor), only `ultra_slow` tests don't. So I don't think that's the issue. It's possible the test used to be slow and isn't anymore.

clamus, 2016-02-03:
It seems @Eric89GXL is right. Doing `nosetests -a '!slow_test'` in my machine gives OK.

clamus, 2016-02-03:
I mean... It used to be `'ultra_slow_test'` and now is '`slow_test`'. Right?

Eric89GXL, 2016-02-03:
Travis excludes ultra slow only, so it's weird that it (a slow test) passes
there but not on your machines. I'll check my machine tomorrow if nobody
has figured it out by then

cle1109, 2016-02-03:
All tests in `test_csp.py` pass on my machine (Mac OS X, Python 3.5.1, all packages up to date; note that I don't use Anaconda).

However, comparing the two covariance estimation methods might not be such a great idea with the example data in the test. It only contains 14 short epochs, so the difference between the filters is not small (that's why I set `decimal=-1`). I think the two variants are only approximately equal with a large number of (longish) epochs.

Anyway, it's still strange that the test works on my machine (and likely also on Travis).

cle1109, 2016-02-03:
Update: I ran the tests with the latest Anaconda Python and all tests pass.

agramfort, 2016-02-03:
works fine on my box...

jasmainak, 2016-02-03:
It fails for @jaeilepp too

Eric89GXL, 2016-02-03:
You all running anaconda? What versions of Python, scipy, numpy, sklearn,
with or without MKL?

clamus, 2016-02-03:
I am running anaconda. My settings are: Python 2.7.11 :: Anaconda 2.4.1 (x86_64), sklearn 0.16.1, numpy 1.9.3, scipy 0.16.1, all with mkl package. Is it a bad idea to keep using python 2.7?

jasmainak, 2016-02-03:
The thing is I can't reproduce it on my work computer which has anaconda. For me, it is Python 2.7.10, Anaconda 1.91 (64-bit), scipy 0.16.0, numpy 1.10.1, sklearn 0.16.1, mkl 1.1.0. Maybe mkl is the difference ...

jaeilepp, 2016-02-03:
Python 2.7.11, anaconda 2.3.0, sklearn 0.17, numpy 1.10.2, scipy 0.16.1, mkl 1.1.0

Eric89GXL, 2016-02-03:
> Is it a bad idea to keep using python 2.7?

I suspect most people here still use 2.7. It's a bad idea to write code that is only 2.7 compatible, but using 2.7 is still fine.

> Maybe mkl is the difference 

I think you can remove MKL by uninstalling the `accelerate` package in conda or something similar, can you try it? If so then you can file a bug with Continuum IO about it.

FWIW I'm using MKL with self-built numpy stack and the test passes here. But I might very well be using a different version of the Intel compilers to build them.

cle1109, 2016-02-03:
I would say it is a good idea to start using Python 3.5.

Since I'm normally not using Anaconda, how do I find out all this information? So far, it looks like the tests fails only on Python 2.7.

jasmainak, 2016-02-03:
I did `conda remove mkl` and now I get [this](https://gist.github.com/jasmainak/de8abca46bbdda413e56) when I run the csp test.

Unfortunately, Mayavi doesn't work with Python 3 :\ To get the information regarding the versions, you can import the package in ipython and do <package>.__version__. For example:

```py
import numpy
numpy.__version__
```

clamus, 2016-02-03:
Ok, I did `conda uninstall mkl`, and then `nosetests mne/decoding/test`, and it works now. It gave a warning (below), but it looks fine.
/Users/lamexicana/repos/mne-python/mne/decoding/time_gen.py:217: UserWarning: Some folds do not have any test epochs.
  warnings.warn('Some folds do not have any test epochs.')

jasmainak, 2016-02-03:
So, it looks like mkl is broken

cle1109, 2016-02-03:
Yes, I can confirm that the comparison fails when `mkl` is installed. @jasmainak I was referring to the Anaconda-specific stuff, but I found it (`conda info` and `conda list`).

clamus, 2016-02-03:
Uhmm... Now, after removing mkl a the DICS test failed ([here](https://gist.github.com/clamus/77bab76da0b9691dba73)).

clamus, 2016-02-03:
Before we continue with the combinatorial issue of "versions". @cle1109 mentioned that the csp test that caused the problem might not make sense for some theoretical reason. Is the solution to change or eliminate the test instead?

cle1109, 2016-02-03:
I wouldn't completely remove the test. A better option would be to test on data with more epochs. I think there are already some tests that require testing data - is this testing data suited for this test?

Eric89GXL, 2016-02-03:
I think there are potentially two problems that need to be fixed. One is that MKL appears to be screwing things up. The other is that the test might not be that robust. So we do need to continue debugging where this comes from with the current test unfortunately.

jasmainak, 2016-02-03:
Yeah, we need to figure out why is mkl screwing up things ...

jasmainak, 2016-02-03:
Looks related: https://software.intel.com/en-us/forums/intel-math-kernel-library/topic/505993

Eric89GXL, 2016-02-03:
Yes perhaps :)

https://github.com/numpy/numpy/pull/6892

Eric89GXL, 2016-02-03:
(you can try locally doing `import scipy; scipy.test('full')` to see if you suffer from the errors, but hopefully Anaconda folks have been using -O1 already)

clamus, 2016-02-03:
@Eric89GXL This is the result of spicy test on my machine
Ran 19617 tests in 595.624s

OK (KNOWNFAIL=148, SKIP=1196)
Out[1]: <nose.result.TextTestResult run=19617 errors=0 failures=0>

Eric89GXL, 2016-02-03:
@cle1109 how about instead of making the test take longer by including more epochs, be more lenient by replacing those two lines with:
```
    for attr in ('filters_', 'patterns_'):
        assert_true(np.corrcoef(getattr(csp, attr).ravel(),
                                getattr(csp_epochs, attr).ravel())[0, 1] > 0.9)
```

Eric89GXL, 2016-02-03:
(we'll still need to solve the MKL issue, but at least this might make the test a bit saner)

Eric89GXL, 2016-02-03:
Also, it's unlikely this is caused by that Fortran bug, since it should have some failures if it was compiled incorrectly. I wonder if it's a `sklearn` bug. Can you try using a different version of `sklearn` (with MKL installed again) to see if it fixes the problem?

cle1109, 2016-02-03:
@Eric89GXL yes, even though such a test for correlation is much weaker, it is still better than no test. Plus, it does work with MKL installed. Bonus: we could increase the correlation threshold to 0.95 (I get a correlation of 0.98), but then the challenge is to fit the whole thing within PEP8-style 79 character lines :-).

Eric89GXL, 2016-02-03:
Where there's a will there's a way:
```
    csp_epochs.fit(epochs_data, y)
    for attr in ('filters_', 'patterns_'):
        assert_true(np.corrcoef(
            getattr(csp, attr).ravel(),
            getattr(csp_epochs, attr).ravel())[0, 1] > 0.95)
```

clamus, 2016-02-03:
Just tested with mkl and sklearn 0.15.0 to 0.17 and every time a CSP test failed:
`conda install mkl`
`conda install scikit-learn=version`
`nosetests mne/decoding/tests/test_csp.py`

jasmainak, 2016-02-03:
@Eric89GXL I get something similar to @clamus . [Here](https://gist.github.com/jasmainak/9c2739213a96ce461bb2) is the output. Not sure what to make of it.

Eric89GXL, 2016-02-03:
There is definitely something strange going on. On Travis the correlation is only 0.74, see:

https://travis-ci.org/mne-tools/mne-python/jobs/106778260

Eric89GXL, 2016-02-03:
I can replicate the bug using conda locally. I'll see if I can figure it out.

Eric89GXL, 2016-02-03:
Looks like the problem is when `proj=True` the `epochs.get_data()` actually differs between MKL and non-MKL (!). I expect this problem to be related to Anaconda using old compilers, see:

https://github.com/ContinuumIO/anaconda-issues/issues/425

I'm going to be mad if this ends up being the issue...

jasmainak, 2016-02-03:
oh boy!

jasmainak, 2016-02-03:
a correlation of 0.95 on your machine but 0.73 on Travis seems to indicate this bug was pretty drastic ...

Eric89GXL, 2016-02-03:
Yes. It turned out to be due to using projections when almost none of the original channels were available. I've fixed it for this particular case for my PR, but we need to address the "incomplete projections" issue before 0.12. I'll track down the issue or if I don't see one, create one.

Eric89GXL, 2016-02-03:
And in this particular case, there were errors around numerical precision that got amplified by the projection operation, so MKL vs OpenBLAS and Anaconda only seemed to play a role in the sense that those numerical errors happened to do less damage (probably randomly) in some cases versus others.

jasmainak, 2016-02-03:
makes sense. Moral is that one can never ignore small numerical errors :)

Eric89GXL, 2016-02-03:
Yep. Time to start doing everything with integers :)

PR #2837 (2016-02-03): MRG: Make test more lenient
Closes #2834.
closed 2016-02-03 by Eric89GXL