mne-tools/mne-python #2840: Error while testing possibly the report module

clamus, 2016-02-03:
Hi Guys, thanks to @Eric89GXL the CSP/mkl fail test is a thing of the past! However, after upgrading to the newest dev version of mne-python, I am getting instead an error while running `nosetests -x` ([here](https://gist.github.com/clamus/cebb0dee459100277081)). Any idea to what this might relate to?

Eric89GXL, 2016-02-03:
I hate to make you traverse the stack up so high, but could you use `nosetests -x --pdb` and python debugger commands `u` (up the stack) `d` (down the stack) and `p <name>` (print variable name) to figure out what the problem is? It's probably most useful if you traverse the stack until you get to this level:
```
  File "/Users/lamexicana/repos/mne-python/mne/viz/misc.py", line 123, in plot_cov
    plt.semilogy(np.sqrt(s) * scaling)
```
Then doing something like `p scaling`, `p s`, or `p np.sqrt(s) * scaling` should give you some insight into whether something is wrong. If there are NaN or inf or negative values in there it could break `matplotlib`'s plotting somehow. They should handle it more gracefully sure, but I suspect we're giving them junk input.

clamus, 2016-02-03:
On the contrary @Eric89GXL , thanks for all the help and support! This is what I got [(here)](https://gist.github.com/clamus/0921f86819cec18782bd). There are no `NaN` or negatives.

clamus, 2016-02-03:
One thing that might be unusual is that the element of `np.sqrt(s) * scaling` in the last row and columns is 3 orders of magnitude smaller than the rest. Might this be related to the issue?

jasmainak, 2016-02-04:
@clamus I can't reproduce this time. Make sure you run `make clean` before running the test. If you can propose a fix, don't hesitate to submit a PR

Eric89GXL, 2016-02-04:
@clamus if you want, you can even do `p np.savez_compressed(...)` to save out the numpy array that causes the failure. Then one of us can try the `plt` command using the exact same dataset. That will tell us if the problem is with your `matplotlib` or the computations that are done before the `plt` call to compute the data.

clamus, 2016-02-04:
@jasmainak I did `make clean` and rerun the test and got the same issue. @Eric89GXL the save output of is ([here](https://www.dropbox.com/s/yql6txpt4guz33z/error_arrays.npz?dl=0)). It was generated via `p np.savez_compressed('error_arrays.npz', s=s, scaling=scaling, scaled_s=np.sqrt(
s)*scaling)`. The element in the last row and column of the scaled array is 3 orders of magnitude smaller. Please let me know if it is a `matplotlib` issue.

Eric89GXL, 2016-02-04:
Works fine over here, so I think your `matplotlib` has a problem:
```Python
import numpy as np
import matplotlib.pyplot as plt
data = np.load('error_arrays.npz')
plt.semilogy(data['scaled_s'])
```
[image]

I'd try installing a different version with Anaconda to see if it fixes the issue.

clamus, 2016-02-04:
Thanks @Eric89GXL ! I will make a clean/new instal of anaconda and try again. That should solve the problem.

closed 2016-02-04 by clamus