mne-tools/mne-python #2847: DOC: connectivity module

dengemann, 2016-02-04:
Public mental note:

- better document the input data API, e.g. how tuples of (array, stc) are combined and what this does to the indices parameter
- document the output API, depending on the method the need to be reshaped. 

For later: write high-level object.

Eric89GXL, 2016-04-20:
Moved to 0.13, feel free to close if you get it done before 0.12 comes out

Eric89GXL, 2016-09-16:
Removing milestone label

AdoNunes, 2019-09-30:
@dengemann Is this still necessary to fix?

dengemann, 2019-09-30:
yes I think nothing has changed.

agramfort, 2019-09-30:
can you clarify what is needed? assuming you want someone else to do the
job ;)

>

dengemann, 2019-09-30:
You should do a for loop over parameters for the connectivity function and see if you understand what it takes to reassemble a proper square matrix. This is what needs to be documented.

mmagnuski, 2021-08-10:
I think this can be closed with connectivity being moving to `mne-connectivity`.

closed 2021-08-10 by mmagnuski