mne-tools/mne-python #2850: BUG: read_raw_brainvision fails for Brain Vision Data Exchange Header File Version 1.0

fraimondo, 2016-02-05:
```Python

/opt/anaconda3/lib/python3.4/site-packages/mne/io/brainvision/brainvision.py in _get_vhdr_info(vhdr_fname, eog, misc, scale, montage)
    350         unit = unit.replace(u'\xc2', u'')  # Remove unwanted control characters
    351         cals[n] = float(resolution)
--> 352         ranges[n] = _unit_dict.get(unit, unit) * scale
    353 
    354     # create montage

TypeError: can't multiply sequence by non-int of type 'float'
```

Digging into it:
```Python
ipdb> unit
'uV'
ipdb> scale
1.0
ipdb> _unit_dict
{'V': 1.0, 'µV': 1e-06}
```

So it looks like 'uV' is not defined in the unit dictionary.

I can make a PR if this is a bug and not a problem of my files. I've got several of them.

jona-sassenhagen, 2016-02-05:
Make a PR I'd say.

Are your BV files created by BV recorder?

fraimondo, 2016-02-05:
Don’t really know where this files came from.

I’ve got the vhdr, vmrk and eeg files.

commit f3fef2bb (2016-02-05): Fix #2850
closed 2016-02-05 by Eric89GXL