mne-tools/mne-python #2852: add more datasets

agramfort, 2016-02-05:
see this email taken from megcommunity mailing list:

In the last year or so, a couple of groups have started making their raw, anonymized, MEEG data available for public download online. Some of these are being added to the http://megcommunity.org/analysis/testdatasets webpage, but here is a rundown of what is out there currently:

-          MMN dataset (http://www.fil.ion.ucl.ac.uk/spm/data/eeg_mmn/ ) used for tutorial/publications applying DCM for ERP analysis using SPM. 

-          Human Connectome Project Datasets (http://www.humanconnectome.org/data/ ). Over a 3-year span (2012-2015), the Human Connectome Project (HCP) scanned 1,200 healthy adult subjects. The available data includes MR structural scans, behavioral data and (on a subset of the data) resting state and/or task MEG data.

-          Kymata Datasets (https://kymata-atlas.org/datasets). Current and archived EMEG measurement data, used to test hypotheses in the Kymata atlas. The participants are healthy human adults listening to the radio and/or watching films, and the data is comprised of (averaged) EEG and MEG sensor data and source current reconstructions.

-          http://www.brainsignals.de/ A website that lists a number of MEG datasets available for download.

we should at least add our MNE datasets in the list of testing data

Eric89GXL, 2017-02-16:
Copied into #2280

Eric89GXL, 2017-02-16:
Oh, you mean we should put our dataset into their list somehow, not that we should expose those datasets?

agramfort, 2017-02-17:
no we should make it easy to pipe in their datasets into mne with fetchers

Eric89GXL, 2017-02-17:
Okay I've added those to #2280

closed 2017-02-17 by Eric89GXL