mne-tools/mne-python #2853: Clean up composite event channel when undesirable and unpredictable overlapping events occur?

jhouck, 2016-02-05:
I'm looking at data from a scan that had absurdly long stimulus markers -- on the order of 500-1000 ms -- with imprecise offsets.  There's no point at which the composite stim channel returns to zero, and in many cases descending events are meaningful.  The stimulus delivery program has been corrected, but occasionally similarly unhelpful things will happen -- buttons get stuck, Presentation/Eprime programming errors occur, etc.  Is there a flexible, easy way to reconstruct what the stim channel should have looked like without the need to do a custom fix for each occasion?  I have an old lisp program for Graph from Matti K. that would do something like this, but that is not really optimal for scripting.  Using find_events with consecutive=True with shortest_event=0 and doing a diff between columns of the resulting array ends up miscategorizing events in cases where the onset of one event coincides with the offset of another.

A small raw file containing only the stim channels can be downloaded from [https://www.dropbox.com/s/tuw33qb67k53ych/show_long_triggers_raw.fif.gz?dl=0] -- you'll see e.g. that although there is an event 17 at sampling point 252541 (494.082 seconds in mne_browse_raw), it would be classified as a 111 on the STI101 data from find_events because it occurred in the sample after a 128 ended.  The next event, an 8 at 252790, is correctly detected, but then there is a spurious event at 252791.  (Note that in the first few minutes of this file there are events from only two categories; jump ahead to the good part).

We routinely collect the old STI001-STI016 along with STI101, so in principle I could use find_events and define_target_events separately with each of these channels to replace the 5's with the appropriate value for the channel (1,2,4,8,16,32, etc), concatenate the resulting arrays, convert the concatenated array into a structured array for sorting, sort, take a diff to detect simultaneous events, sum the event codes for adjacent rows indicated by the diff, swap in the new values, delete the extraneous rows, and so on. I illustrate most of this in a terrible gist at 
[https://gist.github.com/jhouck/c28bbb7a88788936bd99] .  

Does this make any sense or is there a better way?  

Thanks,

Jon

dgwakeman, 2016-02-05:
I actually, intentionally have my events be this long, to ensure that I know precisely how long a given stimulus is presented. This is what the mask is for in find_events. I'm not sure the numbers make sense in your example, as I can't find a 111 instance in the raw file there. It is just that the trigger goes from 128 to 17, indicating that you had a 128 send then changed the value to 17. The easy way to detect this is to just get all your events, by setting consecutive to True. If you do have real examples of combined triggers use the mask option to mask out the uninteresting events.

hth
d

dgwakeman, 2016-02-09:
Did this help at all?

jhouck, 2016-02-09:
Yes, actually -- I switched from using find_events once and using that same array throughout, to using find_events per condition.  More of a derp than an actual issue, thanks for the response.

dgwakeman, 2016-02-09:
@jhouck well, if you switch to consecutive=True, you should be able to avoid running it multiple times. Also, can you close, once you are satisfied with my suggestion

closed 2016-02-09 by jhouck