mne-tools/mne-python #2858: Value error no EOG key in rejection dict

kambysese, 2016-02-07:
`epochs._reject_setup` returns a ValueError because the `eog` value in the `idx` [dict](https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L465) returns an empty list for Vectorview MEEG dataset with 60 EEG channels and EOG61 & EOG62. Debugger shows `idx` contains 60, 102, and 204 elements for `eeg`, `mag`, and `grad` keys. The EOG channels don't appear anywhere in `idx`.

Here is the trace:

jasmainak, 2016-02-07:
you forgot to post the trace

kambysese, 2016-02-07:
Sorry @jasmainak I am editing. Unsure if `epochs` likes the particular rejection dict I am using.

jasmainak, 2016-02-07:
is it possible that you have to set the channel type of your EOG channels? You can use `raw.set_channel_type`

kambysese, 2016-02-07:
It's possible, I've not had to do this before but I also don't recall every using the `eog` arg in rejection dicts, and for some reason `eog=np.inf` is making something unhappy. @Eric89GXL have you ever had to use `set_channel_types` for MEEG data collected at ILABS?

jasmainak, 2016-02-07:
If you don't want to reject based on EOG, just don't add it in the reject dict. If `eog=np.inf` throws an error, it might need fixing though.

Eric89GXL, 2016-02-26:
@kambysese you should not need to use `set_channel_types` unless something went wrong during acquisition. Can you upload the file somewhere on UW servers for me to have a look?

@jasmainak I think right now if the key is in the dict we will throw an error even if it is `np.inf`. I suppose we could change it so that it'll only error if the key is there and finite.

PR #2956 (2016-02-26): MRG+1: Fix reject check
Two minor fixes in one PR to (hopefully) save reviewer and CI time.

Closes #2858.
Closes #2855.
closed 2016-02-26 by Eric89GXL