mne-tools/mne-python #2859: Importing CTF data on machine with different locale.

nwilming, 2016-02-08:
I'm trying to import a CTF data set that has been recorded with locale en_US on my local machine which has de_DE. This fails because a date string can't be parsed:
```
>>> locale.setlocale(locale.LC_ALL, 'de_DE')
>>> mne.io.read_raw_ctf('20151208_01.ds')
ds directory : /Volumes/dump/conf_meg/raw/s01-01_Confidence_20151208_01.ds
    res4 data read.
    hc data read.
    Separate EEG position data file read.
    Quaternion matching (desired vs. transformed):
       2.12   82.38    0.00 mm <->    2.12   82.38    0.00 mm (orig :  -49.12   66.08 -256.99 mm) diff =    0.000 mm
      -2.12  -82.38    0.00 mm <->   -2.12  -82.38    0.00 mm (orig :   52.98  -63.28 -255.14 mm) diff =    0.000 mm
      98.83    0.00    0.00 mm <->   98.83   -0.00   -0.00 mm (orig :   77.27   64.28 -244.34 mm) diff =    0.000 mm
    Coordinate transformations established.

---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-6-e495908c73ec> in <module>()
      1 locale.setlocale(locale.LC_ALL, 'de_DE')
----> 2 mne.io.read_raw_ctf('20151208_01.ds')

/Users/nwilming/src/mne/mne/io/ctf/ctf.pyc in read_raw_ctf(directory, system_clock, preload, verbose)
     60     .. versionadded:: 0.11
     61     """
---> 62     return RawCTF(directory, system_clock, preload=preload, verbose=verbose)
     63 
     64 

/Users/nwilming/src/mne/mne/io/ctf/ctf.pyc in __init__(self, directory, system_clock, preload, verbose)

/Users/nwilming/src/mne/mne/utils.pyc in verbose(function, *args, **kwargs)
    549         with use_log_level(verbose_level):
    550             return function(*args, **kwargs)
--> 551     return function(*args, **kwargs)
    552 
    553 

/Users/nwilming/src/mne/mne/io/ctf/ctf.pyc in __init__(self, directory, system_clock, preload, verbose)
    109         coord_trans = _make_ctf_coord_trans_set(res4, coils)
    110         # Compose a structure which makes fiff writing a piece of cake
--> 111         info = _compose_meas_info(res4, coils, coord_trans, eeg)
    112         # Determine how our data is distributed across files
    113         fnames = list()

/Users/nwilming/src/mne/mne/io/ctf/info.pyc in _compose_meas_info(res4, coils, trans, eeg)
    370     info['meas_id']['usecs'] = 0
    371     info['meas_id']['secs'] = _convert_time(res4['data_date'],
--> 372                                             res4['data_time'])
    373     info['experimenter'] = res4['nf_operator']
    374     info['subject_info'] = dict(his_id=res4['nf_subject_id'])

/Users/nwilming/src/mne/mne/io/ctf/info.pyc in _convert_time(date_str, time_str)
     69             break
     70     else:
---> 71         raise RuntimeError("Illegal date: %s" % date)
     72     for fmt in ('%H:%M:%S', '%H:%M'):
     73         try:

UnboundLocalError: local variable 'date' referenced before assignment
```
Setting the locale to en_US (`locale.setlocale(locale.LC_ALL, 'de_DE')`) fixes this. 

Maybe `_convert_time` in mne.io.ctf.info could spit out an appropriate error message that points out that the locale needs to be set correctly? I assume that the locale of the data will usually be en_US, so a helpful error message could be:

"Illegal date %s.  If the language of the data does not correspond to your local machine's language try to set the locale to the language of the date string: locale.setlocale(locale.LC_ALL, 'en_US')"

I also think that the line 71 in info.py should be: `raise RuntimeError("Illegal date: %s" % date_str)` and not reference the variable `date` as it does not exist yet.

I'm running mne version 0.12.dev0.

jasmainak, 2016-02-08:
+1 and PR welcome

agramfort, 2016-02-08:
PR welcome or share a file so we can replicate.

commit f1d32d09 (2016-02-13): FIX #2859: Add informative error message when date string can not be parsed
closed 2016-02-13 by agramfort