mne-tools/mne-python #2860: ENH/API: new contracts and conditions for mne.io.meas_info.Info

dengemann, 2016-02-08:
The recent discussion in #2765 has lead me to discuss with Eric about different ways in which we could improve the Info API to be more clearer. One aspect that maybe was not so clear during the discussion we had over there was the notion of semantic VS code complexity. My idea was that we can have complex implementations but clear APIs that you can quickly conceptualise without exceptions and sub-rules. As it looks now we would have to say: 

"our info is like a dict but you cannot use the dict API without constraints, as it has some extra behavior, depending on the keys it will be read only and it requires a subclass of list with reduced API for ch_names"

Maybe it's time to change the API and introduce explictly a new object with simplified semantics

"our info is a read only dict with nice convenience extras, but items cannot be set directly after construction."

it would then allow for a worflow like this:

```Python

info_as_dict = dict(info)
info_as_dict['lowpass'] /= 2
info = Info(info_as_dict)

```
This could then be the internals of a public or private _update method that takes keyword arguments and hence allows for easier consistency and value checking. It could return a copy to be consistent. We could also use the [FrozenDict pattern](http://stackoverflow.com/questions/2703599/what-would-a-frozen-dict-be)

It would also allow us to implement expose a save method etc.

We could then send a clear message with the release and have a nice documentation that prepares everyone for this backwards incompatible change (without deprecation)

@wmvanvliet @Eric89GXL @jasmainak @agramfort, @jona-sassenhagen ..., ...

Let me know what you think.

Eric89GXL, 2016-02-08:
@wmvanvliet pointed out in the other thread that we need these to work:
```
info['bads'] = [...]
info['bads'] += [...]
```
But I think those are the only direct-write methods that we'd need to support. So we could say it's frozen other than the `bads` attribute. One advantage to making the `bads` setting a little bit smarter is that we can do things like emit warnings when bads are added that exist in projs. So we'd get that as an advantage, too.

Are there other usage patterns that wouldn't work making it write-only? I'm not worried about how we modify things internally in private functions (e.g., in `set_channel_types`), but rather in terms of what users would actually access and set. `info['bads']` is all that comes immediately for mind to me.

dengemann, 2016-02-08:
I think it might also be a good moment to abandon this, it is not very explicit anyways. 
Why not have a `.set_bads(ch_names)` method. Also we could have other setters (that would always return a copy).

wmvanvliet, 2016-02-09:
Before I can weigh in on this, I need to be educated a little.

Can someone explain to me why some data structures are implemented as dicts (info, forward) and others as full classes (source estimate, montage)? I always thought this was just historical, but @dengemann hinted at deeper reasons of having simple, flat objects. If I were to implement a new datastructure, should it be a dict or a class?

dengemann, 2016-02-09:
The idea is that dicts are nice general purpose heterogenous data containers. So if the purpose of a class is essentially to contain data we're using dicts. If the class is also there to process data (includig viz), we most of the time don't subclass dicts. I think that's the main distinction.

wmvanvliet, 2016-02-09:
ok, that makes sense. thanks.

wmvanvliet, 2016-02-09:
Here are MNE's dict structures:

- `ConductorModel`
- `Covariance`
- (`_DecodingTime`)
- `Forward`
- `Info`
- `InverseOperator`
- `Projection`

Most if not all of them have some consistency requirements. For example, the length of the `ch_names` field should be the number of rows in the `data` field. In cases like this, the immutable pattern makes a lot of sense. New objects can be created, but afterwards, never altered. If a value needs altering, you make a new object, optionally using the old object as a template. The main advantage of this approach is that you can always have sane consistency checking.

Making these objects immutable would mean that instead of subclassing `dict`, they would subclass `collections.Mapping`. Or alternatively subclass a `FrozenDict` which we would make ourselves. Any data stored inside the structure would also have to be immutable objects. All primitive types (int, str, float, bool) are immutable, so this mainly means that any dicts will become `FrozenDict`'s and any lists will become tuples. (If we would not do this, things like `info['bads'] += 'foo'` would still work and bypass the consistency checks.)

Obviously, this would require many small refactorings across our code base.

IMO, the main question we must ask ourselves is how much user script we're going to break with this. How common is it for a user to mess with the values in a dict structure?

dengemann, 2016-02-09:
I disagree that the subtypes have to be frozen too. My point was to have
clear API pattern instead of write protecting some fields but not others.

agramfort, 2016-02-09:
in the past dict were converted to object eg to add a nice __repr__ or to
add read/write methods

jaeilepp, 2016-02-09:
With this new ``ChannelNameList`` the output of ``raw.info['ch_names']`` is quite uninformative + it throws a bunch of warnings for me:
```
/home/jaakko/anaconda/lib/python2.7/site-packages/IPython/core/formatters.py:92: DeprecationWarning: DisplayFormatter._ipython_display_formatter_default is deprecated: use @default decorator instead.
  def _ipython_display_formatter_default(self):
/home/jaakko/anaconda/lib/python2.7/site-packages/IPython/core/formatters.py:669: DeprecationWarning: PlainTextFormatter._singleton_printers_default is deprecated: use @default decorator instead.
  def _singleton_printers_default(self):
/home/jaakko/anaconda/lib/python2.7/site-packages/IPython/core/formatters.py:672: DeprecationWarning: PlainTextFormatter._type_printers_default is deprecated: use @default decorator instead.
  def _type_printers_default(self):
/home/jaakko/anaconda/lib/python2.7/site-packages/IPython/core/formatters.py:677: DeprecationWarning: PlainTextFormatter._deferred_printers_default is deprecated: use @default decorator instead.
  def _deferred_printers_default(self):
Out[3]: <ChannelNameList | 320 channels | IASX+, IASX-, IASY+, IASY-, IASZ+ ... MEG2642, MEG2643, STI101, SYS101, SYS201>
```
This should be fixed, unless we decide to roll back.

wmvanvliet, 2016-02-09:
@jaeilepp how would you like the output to look?

wmvanvliet, 2016-02-09:
> I disagree that the subtypes have to be frozen too. My point was to have clear API pattern instead of write protecting some fields but not others.

These two statements sound contradictory to me. You want all the fields to be write protected, but you only want to do this for `Info` dicts and not other dicts where it would make just as much sense.

agramfort, 2016-02-09:
> @jaeilepp <https://github.com/jaeilepp> how would you like the output to
look?

like before :)

dengemann, 2016-02-09:
> You want all the fields to be write protected, but you only want to do this for Info dicts and not other dicts where it would make just as much sense.

I did not comment on that actually. I only talked about the values of that dict that do not have to be frozen necessarily. Well we can just keep the dict API in many other cases or we could also use kind of a frozen pattern.

wmvanvliet, 2016-02-09:
here is an alternative proposal:

we revert #2765 and keep the dict structures as they are, but we add to each of them a public `update` method and a private `_check_consistency`. We make it clear in the docs that the `update` method is the preferred way to modify the data structure. But we don't strictly enforce this, hence, no existing code is broken at all. If in the future a user doesn't use the `update` function, it's his/her own fault if things get buggy ;)

dengemann, 2016-02-09:
That sounds like a reasonable direction. Subclassing the update method would allow us to do all sorts of checking on the fly. And the API pattern would be simple such that we could easily alert users about it in the next release. We could still freeze the `__set_item__` methods if that would make us feel safer.

dengemann, 2016-02-09:
... we could of course also subclass the item setters and issue warnings telling people to use the update method.

wmvanvliet, 2016-02-09:
since it's difficult to perfectly overwrite item setters perfectly anyway, I'd go for warnings, yes.

wmvanvliet, 2016-02-09:
could we keep the automatic updating of `nchan` and `ch_names` when we call `update` or use the constructor of `Info`? It really saves quite a bit of redundant code.

dengemann, 2016-02-09:
Yes that is the idea. I would say like this it is very clean and simple.

dengemann, 2016-02-09:
Independently of this we might want to introduce instance.set_bads(ch_names)

jona-sassenhagen, 2016-02-09:
Hm ... shouldn't there be, ideally, one, and just one, way of doing something?

wmvanvliet, 2016-02-09:
in language design and a standard library, yes. But in an analysis package like MNE where a substantial amount of work happens through the REPL? I'm not so sure. Take `pandas` for example, it has many ways to accomplish the same thing.

wmvanvliet, 2016-02-09:
and in MNE, we offer multiple paths all over the place. `mne.viz.plot_evoked` versus `evoked.plot` anyone?

dgwakeman, 2016-02-09:
IMO (and I realize I am more user than developer) these examples are bugs in MNE not features. I know breaking API sucks. Maybe, these are lessons to be learned about how new features should be developed, maybe do a post-mortem, on why these things were done this way to try to help us learn how not to in the future.

wmvanvliet, 2016-02-09:
Well then, `raw.info.update(bads=['EEG010', 'EEG040'])` obviously going to be a way to set bad channels. So thats a no for `raw.set_bads(['EEG010', 'EEG040'])` then?

dengemann, 2016-02-09:
Yeah that's right. Although an instance based method would be easier to read. 

raw.set_bads(ch_names, mode='overwrite'|'append')

jaeilepp, 2016-02-10:
>> @jaeilepp <https://github.com/jaeilepp> how would you like the output to
>>look?

>like before :)

Yeah exactly. It used to just print the names.

wmvanvliet, 2016-02-10:
Shall we go ahead and revert #2765 then?

dengemann, 2016-02-10:
> Shall we go ahead and revert #2765 then?

we would keep the consistency checking but go for warnings on set item and overwriting the update method?

wmvanvliet, 2016-02-10:
> we would keep the consistency checking but go for warnings on set item and overwriting the update method?

That would have my +1. But we can still discuss this further. In the meanwhile, it doesn't seem worth it to continue working on  tweaks to #2765 if we're going to revert it anyway. And if we're going to revert it, maybe do it sooner than later?

jona-sassenhagen, 2016-02-10:
@dengemann are you even in favor of revertig?

dengemann, 2016-02-10:
Welll it's not reverting is it, it is pulling out a few things and exposing them more generally and continuing development on the API issue that was discovered by #2765. Let's just make sure people agree. cc @jaeilepp @agramfort @jasmainak @Eric89GXL ...

Eric89GXL, 2016-02-10:
@dengemann what is the proposal now? From the issue description it sounds like it is to to make the `Info` dict frozen, but now it sounds like the discussion has shifted to just making some/more fields read-only...?

agramfort, 2016-02-11:
what's the summary here? sorry for pitching in so late

dengemann, 2016-02-11:
The idea was that we should raise for each dict object warning on updating the fields via item setters and implement a controlled .update method that takes care of all consistency checking. No locked fields. Clear documentation on these sort of containers and encouraging the use of the .update method. Keep automatic updates of `nchan` but no locked keys. Was that it @wmvanvliet ?

wmvanvliet, 2016-02-12:
yes!

Eric89GXL, 2016-02-15:
sounds reasonable to me

Eric89GXL, 2016-04-14:
We should really resolve this one for 0.12 so we don't immediately need to deprecate. @wmvanvliet do you have any time you could dedicate to this in the next week or two?

wmvanvliet, 2016-04-15:
I'll see what I can do. Any help is appreciated! :)

Eric89GXL, 2016-04-15:
One thing we could do is more or less revert the changes -- make a single commit that changes it back to deriving from `dict`. Then we can deal with the more complex solution later. WDYT? If so I might be able to do the first "fix".

Eric89GXL, 2016-04-20:
@wmvanvliet I assume by the silence that you will be too busy to pursue something comprehensive in the next week or two for 0.12, so I'll go ahead with the revert-like plan tomorrow or the next day unless I hear otherwise.

wmvanvliet, 2016-04-20:
That would be best. I don't have the time right now to make such an invasive change properly. I will pick this up though :) Maybe for 0.13.

Eric89GXL, 2016-09-08:
Can we close this one for now unless someone is really interested in enhancing the object?

agramfort, 2016-09-08:
+1 to close

Eric89GXL, 2016-09-08:
We can reopen if someone gets motivated again

closed 2016-09-08 by Eric89GXL