mne-tools/mne-python #2862: ENH: Investigate shared memory writing in parallel computations

Eric89GXL, 2016-02-08:
There are many places in our code where we effectively use the pattern:
```
out = np.array(parallel(...))
```
But this is painful from a memory standpoint because we end up with two copies of the result in memory wher doing the `np.array(...)` part of that command. This can be alleviated by using memmapping:

https://pythonhosted.org/joblib/parallel.html#writing-parallel-computation-results-in-shared-memory

I'd expect this to be useful e.g. in something like `raw.filter` and our time-frequency functions. Finding places we've done `if n_jobs == 1` will suggest this might be useful.

At the input end, we can also perhaps save (a lot!) of memory by also using memmapping:

https://pythonhosted.org/joblib/parallel.html#working-with-numerical-data-in-shared-memory-memmaping

This could lead to additional memory savings by not sending `N` copies of the data to workers. I'd expect this to help e.g. clustering methods, but I think we already have it there. There are probably other places it would be useful, too.

Or have we tried doing this before and I just don't remember?

dengemann, 2016-02-08:
> Or have we tried doing this before and I just don't remember?
 @Eric89GXL

We have support for this in our stats function for spatiotemporal clustering. Tried it once but it was not something that turned out too relevant for me.

agramfort, 2016-02-08:
AFAIK we never tried

just see that it means writing tmp files to disk. This can be a pb if there
is too many

jasmainak, 2016-02-08:
I think it can lead to race conditions and won't work reliably.

jasmainak, 2016-02-08:
Or are you thinking or writing one file for each process? Then yeah, the tmp fills up. I once had an issue where my computer took 20 minutes to boot because I had filled up the tmp folder with joblibs stuff :\

Eric89GXL, 2016-02-09:
> I think it can lead to race conditions and won't work reliably.

According to their docs, as long as you write to non-overlapping bits of the memmapped array, it should be fine. In other words, doing the equivalent of `np.sum(parallel(...))` isn't expected to work, but doing the equivalent of `np.array(parallel(...))` should. Have you tried using it this way and had problems previously?

larsoner, 2018-08-18:
Closing this one until we find a use case where we need it

closed 2018-08-18 by larsoner