mne-tools/mne-python #2864: Memory consuming functions

jaeilepp, 2016-02-09:
I ran memory profiler on our examples to track down memory consuming functions. I thought that I'd share the results here if someone else wants to look at it. The examples listed here have a peak consumption of over 1 GiB.
```
datasets/plot_brainstorm_data.py
datasets/plot_spm_faces_dataset.py
decoding/plot_decoding_spatio_temporal_source.py
decoding/plot_decoding_time_generalization.py
forward/plot_source_space_morphing.py
inverse/plot_covariance_whitening_dspm.py
inverse/plot_tf_lcmv.py
preprocessing/plot_resample.py
simulation/plot_simulate_raw_data.py
stats/plot_linear_regression_raw.py
time_frequency/plot_compute_raw_data_spectrum.py
time_frequency/plot_epochs_spectra.py
visualization/plot_meg_eeg_fields_3d.py
```

agramfort, 2016-02-09:
do you have the exact peak value for all of them?

jaeilepp, 2016-02-09:
Not right now, but I can check.

PR #2977 (2016-03-10): MRG: Better memory usage (#2)
More memory tweaks.

Closes #2864.
closed 2016-03-10 by agramfort