mne-tools/mne-python #2866: ENH/IO?: ICAArray

dengemann, 2016-02-10:
Sometimes it is important to to read ICA solutions from different packages. To simplify this we probably need some new abstraction. I still have to think about it but the idea would be to write an ICAArray class or a function that simplifies this tasks by accepting unmixing matrices, preprocessing weights, etc. scalings etc. and whatever and returns a ready-to-use ICA object.

cc @jasmainak @jona-sassenhagen @jmontoyam

dengemann, 2016-02-10:
cc @agramfort

jasmainak, 2016-02-10:
the two AA in ICAArray hurt my eyes :)

dengemann, 2016-02-10:
> the two AA in ICAArray hurt my eyes :)

I thought the same.

jona-sassenhagen, 2016-02-10:
Also, maybe we want to keep it open to other forms of decomposition/BSS..?

dengemann, 2016-02-10:
> Also, maybe we want to keep it open to other forms of decomposition/BSS..?

yes good idea as long as we can assume a common API.

Maybe `DecompositionArray`?

jona-sassenhagen, 2016-02-10:
`SpatialFilterArray`?

Although that in principle should include e.g. CSP.

dengemann, 2016-02-10:
Maybe simply `SpatialFitler` or `Decomposition` as we don't store data here. CSP is a bit different thought, it is, so to say, a cross-decomposition.

agramfort, 2016-02-10:
Why is the current ICA object not enough?

dengemann, 2016-02-10:
> Why is the current ICA object not enough?

it might be enough, but maybe we could have a constructor function that takes the essential data and sets all the required attributes. Imagine you reade an ICA that does not use PCA etc. so you will be happy if the function sets for you the pca attributes to identity. Another potential issue might be related to the ordering of channels, etc. Can you feel where this is going?

jona-sassenhagen, 2016-02-10:
I see it like Denis - it would be useful to have a constructor that can create proper Ica objects from arbitrary (un)mixing matrices.

agramfort, 2016-02-11:
let's see if we can make it work sticking to a single ICA object.

dengemann, 2016-02-11:
I think it will work, but we need a function.

dengemann, 2016-02-11:
`assemble_ica`

`construct_ica`

....

agramfort, 2016-02-11:
draft something. We'll argue about the name after :)

dengemann, 2016-02-11:
Yeah will draft something based on a current problem I want to solve.

jona-sassenhagen, 2016-03-07:
This one still on the menu?

dengemann, 2016-03-07:
Not that easy, dev time is rather high if you want to get it right. Foreign ICAs can be very different ...

Eric89GXL, 2016-05-11:
Given the technical difficulties, should we close?

dengemann, 2016-05-11:
Yes for now.

closed 2016-05-11 by Eric89GXL