mne-tools/mne-python #2867: compute_raw_covariance vs compute_covariance

Eric89GXL, 2016-02-10:
Looking at `compute_raw_covariance`, it is not as good as `compute_covariance` because it can't make use of the advanced regularization methods. Looking at the code, it seems like we could have `compute_raw_covariance` create an `Epochs` instance under the hood using `make_fixed_length_events` and then call `compute_covariance`. @dengemann I think we discussed doing this before, is there some reason we decided not to do it that way?

dengemann, 2016-02-10:
>  @dengemann I think we discussed doing this before, is there some reason we decided not to do it that way?

Cost-benefit analysis suggests that we won't win so much here, as regularization is less important for estimates from raw data. On the other hand in our PRNI paper we showed how our improved scaling defaults implement a second security layer by regularising via rank-estimation on the SVD.

jona-sassenhagen, 2016-02-10:
Is there a function to split up a raw object into a series of evenly spaced, consecutive, possibly overlapping epochs? Because if no, and if you decide on implementing such functionality in the context of this, it would be good to make such a function.

dengemann, 2016-02-10:
> Is there a function to split up a raw object into a series of evenly spaced, consecutive, possibly overlapping epochs?

You are looking for `mne.event.make_fixed_length_events`

FYI we already had a darft for calling the cov epochs function under the hood, it was computationally not really tractable.

Eric89GXL, 2016-02-10:
I actually took a stab at implementing it, there is only one small detail that differs. For `raw`, the last window gets included even if it's too short. This seems more like a bug than a feature since not all the window lengths are the same. Do we really need to have it in there? I guess it makes the result identical to what the C code gives, but I'm not sure it's worth it.

> Cost-benefit analysis suggests that we won't win so much here, as regularization is less important for estimates from raw data.

That is true assuming you have a lot of raw data to use to compute the epochs, right? Seems like this would only be true for certain combinations of raw durations and `step` arguments. In other words, if you wanted to use long windows like ~10 seconds and ~5 minute file, you might actually want regularization. Or if you only had a minute of data and wanted to use 1 second windows, etc.

The code is also quite a bit simpler with the changes to using `make_fixed_length_events` and `Epochs`.

PR #2868 (2016-02-24): [MRG+1]: Unify Raw and Epochs-based covariance
Closes #2867. Related to #2448.

This simplifies our code a bit, and also shows how `compute_raw_covariance` differed from `compute_covariance` under the hood.

As shown in the code, `compute_raw_covariance` has (effectively) `keep_sample_mean=False` by default, whereas `compute_covariance` has `keep_sample_mean=True` by default. This makes me think we should have `keep_sample_mean=None` -> `False` with deprecation message saying it will change to `True` in 0.13. This would really unify the two functions. Thoughts?
closed 2016-02-24 by Eric89GXL