mne-tools/mne-python #2869: maxwell filter ctc and cal file

kambysese, 2016-02-11:
I am working with vectoview data where a sensor triplet was turned off. `maxwell_filter` errors out because the number of channels in the fif file do not match the numbers in the fine calibration and ctc files. Perhaps preserving the behavior of Maxfilter program and treating the system as if only N channels as per the number in the recorded dataset?

Eric89GXL, 2016-02-11:
It should probably be able to sub-select channels. But I'm not sure if the cross-talk matrix needs to be modified somehow to compensate. @staulu do you know?

Can you share the file, possibly using `raw.anonymize()` or `mne_anonymize`? You can try `raw.anonymize().crop(0, 1).save('test_missing_raw.fif', buffer_size_sec=1.)`, it might do the trick.

kambysese, 2016-02-11:
Find file [here](https://www.dropbox.com/s/v8s74kwks06zv3s/test_missing_raw.fif?dl=0). 
I tried rm lines in cal file and that worked but @staulu mentioned that modifying the ctc file "would be more tricky" :)  Samu did suggest the files be used during maxwell filtering as though the system is 30x instead of 306 channels, and also that this could be a regular occurrence and best to preserve the MF behavior.

staulu, 2016-02-11:
To be exact, the missing channels are physically there, even if they have been excluded from the channel list. This means that they do slightly interfere with some of the neighboring channels through mutual inductance etc. Therefore, their contribution should be taken into account by the CTC matrix. But if we only have data available from, e.g., 303 channels, the next best thing is to exclude the cross-talk columns and rows corresponding to the missing channels. I don't expect this to be a big problem if there are just a few missing channels. 

Samu

Eric89GXL, 2016-02-12:
Ahh good. That's an easy solution to implement :)

closed 2016-02-15 by agramfort

kambysese, 2016-02-16:
@Eric89GXL new trace....

>   File "/home/ktavabi/Projects/mne-python/mne/utils.py", line 596, in verbose
>     return function(*args, **kwargs)
>   File "/home/ktavabi/Projects/mne-python/mne/io/base.py", line 1284, in save
>     drop_small_buffer, split_size, 0, None)
>   File "/home/ktavabi/Projects/mne-python/mne/io/base.py", line 1942, in _write_raw
>     reset_range)
>   File "/home/ktavabi/Projects/mne-python/mne/io/base.py", line 2085, in _start_writing_raw
>     write_meas_info(fid, info, data_type=data_type, reset_range=reset_range)
>   File "/home/ktavabi/Projects/mne-python/mne/io/meas_info.py", line 1248, in write_meas_info
>     write_ch_info(fid, c)
>   File "/home/ktavabi/Projects/mne-python/mne/io/write.py", line 325, in write_ch_info
>     fid.write(np.array(ch['loc'], dtype='>f4').tostring())  # writing 12 values
> KeyError: 'loc'

PR #2883 (2016-02-15): MRG: Allowing missing chs in maxwell filter
Closes #2869.