mne-tools/mne-python #2870: DeprecationWarning in pandas after importing mne

cle1109, 2016-02-11:
Whenever I

    import mne
    import pandas as pd

I get the following warning:

```
/Users/test/anaconda/lib/python3.5/site-packages/pandas/__init__.py:7: DeprecationWarning: bad escape \s
  from pandas import hashtable, tslib, lib
```

Importing `pandas` in a clean session works. This is a fresh Anaconda install where I added MNE via `pip install mne`.

dengemann, 2016-02-11:
not an MNE issue ... maybe different pandas versions on your system?

cle1109, 2016-02-11:
I just installed a fresh Anaconda on a different Windows machine and I get the same warning. I only see this message if I first import `mne`. So MNE is at least part of the problem.

agramfort, 2016-02-11:
mne is maybe affecting a global setting on warnings ...

is this recent on mne master?

Eric89GXL, 2016-02-11:
Oh right. Yes we do:

https://github.com/mne-tools/mne-python/blob/master/mne/utils.py#L463

Eric89GXL, 2016-02-11:
(and we've had that line for a *long* time, so it's more likely that Pandas has some recent change that now elicits a DeprecationWarning than it is that we changed something)

cle1109, 2016-02-12:
So - is changing the warning state necessary? And I don't quite understand the warning anyway, what's wrong with the Pandas code? Should they change anything?

Eric89GXL, 2016-02-12:
It's necessary if we want users to see Deprecation messages, which are pretty useful. So I think we should keep it. The warning is probably due to some change in Pandas code, but they might not see they are using some deprecated behavior because they don't have them enabled by default.

Eric89GXL, 2016-02-12:
...but before bothering them about it, you might want to verify that replacing `import mne` with some suitable `warnings` call before the `import pandas` does indeed elicit the same warning.

cle1109, 2016-02-15:
Yes. The following minimal example triggers the warning:

    import warnings
    warnings.simplefilter("default")
    import pandas as pd

I'll let the Pandas devs know.

However, I'm still not convinced that changing the warnings filter is a good idea. As we saw, it might trigger a whole lot of warnings in other packages. Since DeprecationWarnings are ignored by default, there must be a way to show these to the user other than changing the global warnings filter (I think scikit-learn is showing DeprecationWarnings to users, but without changing the global warnings state).

Eric89GXL, 2016-02-15:
Indeed:

```
warnings.filterwarnings('always', category=DeprecationWarning,
                        module='^{0}\.'.format(re.escape(__name__)))
```
I'll open a PR

PR #2880 (2016-02-15): MRG: Fix warning filtering
Closes #2870.
closed 2016-02-15 by agramfort