mne-tools/mne-python #2871: Plot the sensor array or layout of MEG/EEG data

teonlamont, 2016-02-11:
It would be super useful if there were a simple viz plotting function for showing the sensor array. It would just be the layout plot that we currently have but it would be without data and labels for the different sensors

Eric89GXL, 2016-02-11:
I wonder if we can integrate this with the existing function somehow. Maybe add a `title_color=None` kwarg that you could set to e.g. `'w'` that would overlay the channel name above the plotting axis like a small title. Then to get what you want, you could just set the channel colors to zero for an `evoked` file, but we also get some nicer functionality when there is data present. WDYT?

jona-sassenhagen, 2016-02-11:
This is one of the functions I'm constantly patching together on the fly myself. The necessary components are in the MNE code, but there should be a convenience function.

jasmainak, 2016-02-11:
yeah, we should have this function.

agramfort, 2016-02-11:
inst.plot_sensors_3d ?

jona-sassenhagen, 2016-02-11:
Intuitively, I'd like to see only one function (plot_layout? plot_montage?) with a keyword to set 2d vs 3d.

agramfort, 2016-02-11:
good idea

jasmainak, 2016-02-11:
yes, +1 :)

dengemann, 2016-02-11:
I had something like this somewhere ...

dengemann, 2016-02-11:
Well anyways it's simple a few lines of matplotlib. My +1 too.

closed 2016-04-14 by dengemann