mne-tools/mne-python #2872: Make event_id = None by default in constructor

teonlamont, 2016-02-11:
It seems like the default behavior for specifying the `event_id` should be None. Anybody opposed to setting a default. It would remain backwards-compatible.

Eric89GXL, 2016-02-11:
You'll need to also set `tmin` and `tmax` to something because they come afterward, assuming you're talking about the `Epochs` constructor. Would you propose to have it be `tmin=None, tmax=None` and throw an error if people don't set them? I guess then people could do `Epochs(raw, events, tmin=-0.2, tmax=1.)` instead of `Epochs(raw, events, None, -0.2, 1.)`. Is that the idea?

jona-sassenhagen, 2016-02-11:
I'd like to see defaults (None, -.2, 1) for event_id, tmin, and tmax. A backwards-compatible change that makes a lot of stuff less verbose.

agramfort, 2016-02-11:
>
> I'd like to see defaults (None, -.2, 1) for event_id, tmin, and tmax. A
> backwards-compatible change that makes a lot of stuff less verbose.
>
fine with me although I would use tmax=0.5 as default :)

jona-sassenhagen, 2016-02-11:
I'd just make sure to use something that doesn't break your computer if you happen to have 1000+ events sampled at 1000 Hz. (I guess `preload=False` by default, but still.)

agramfort, 2016-02-11:
let's set defaults then

any volunteer?

kingjr, 2016-02-11:
I'm also +1 for defaults. I would recommend (0, 1) for tmin tmax, at least it'll be in percentage of the original time.

jona-sassenhagen, 2016-02-11:
0, 1 doesn't play well with baselining though does it? Also I *really* hate it when people don't show their baseline in their plots.

kingjr, 2016-02-11:
If they want to baseline, they should be required to set the timing manually. 

> Also I really hate it when people don't show their baseline in their plots.

ha!

kaichogami, 2016-02-21:
Hello,
I would like to work on this.

dengemann, 2016-02-21:
Sorry overlooked this one. I'm not really convinced we need these defaults here. You should really set them explicitly. Ok for `event_id` which would just mean: take all events. But tmin and tmax should really be chosen explicitly. People should be reminded of this by not allowing them to construct epochs without making a choice, so her'es my -1 on this. Do we *really* need this?

jasmainak, 2016-02-21:
I agree with @dengemann. `tmin` and `tmax` should be set explicitly. But `event_id` can be set to `None` by default, which would mean -- take all events.

jona-sassenhagen, 2016-02-21:
If we set event_id to a kwarg, but not tmin and tmax, we're changing the order of the args, don't we?

I am not convinced defaults for tmin and tmax are inherently bad either. "People should be reminded"? This is not a thing where a bad default will slip in unexpectedly; either your epoch time window shows you what you care about, or it doesn't, and it is immediately clear if it does. This is different from stuff that can silently creep in, like baselining (which we *do* have a default for!).

We do, however, not strictly need it. In analyzing real data, you will almost always decide on a specific time window. The only real difference would be in examples and tests, where it allows you to be a bit more concise.

In summary, I'm not convinced of the arguments against defaults for tmin and tmax, but I'm not strongly pro either.

dengemann, 2016-02-21:
Maybe it's just not worth the effort then. Changing the API for just some comfort is not necessary.

jona-sassenhagen, 2016-02-21:
It's a two-line change though that in the long run could save hundreds of lines for demos and tests.

jona-sassenhagen, 2016-02-21:
But am I seeing it correctly that if we want to keep the current API, we need to make either all 3 kwargs (due to the order), or keep them all as positional args?

dengemann, 2016-02-21:
> we need to make either all 3 kwargs (due to the order), or keep them all as positional args?

That's right yes. 

> It's a two-line change though that in the long run could save hundreds of lines for demos and tests.

Another good argument for the tests. For the demos this exactly not what we want to keep implicit.

Eric89GXL, 2016-02-21:
We could make tmin=tmax=None (raise error) and event_id=None (use all) by
default. Then people could at least do

```
Epochs(raw, events, tmin=-0.2, tmax=1.0)
```

But that isn't all that elegant.

Thinking about it more, the -0.2 at least does make sense as a default
since it's very widely used. At this point I'm +0.5 on setting tmin/tmax to
some numerical defaults. It seems like it could simplify a few things,
especially when someone just wants a quick minimal look at some data.

dengemann, 2016-02-21:
Ok I buy the arguments for simplifying tests. But if we add these defaults let's add some verbosity and keep explicit setting in examples.

jona-sassenhagen, 2016-02-21:
One way to make it verbose would be along these lines:

```
...
Epochs(inst, events, event_id=None, tmin=None, tmax=None, ...)
...
    if tmin == None:
        warnings.warn("No window start giving, setting to -.2.")
        tmin = -.2
```

etc

But I'm not sure that's worth it.

kingjr, 2016-02-21:
> `warning.warn("No window start giving, setting to -.2.")

warning.warn("No tmin given, defaults to -.2.")

And since we're there, we can also make a default `events` too, can't we?

jasmainak, 2016-02-21:
One of the things I would really love to see is a `raw.to_epochs()` method. Basically, here you can use all the `events` in `raw.events` and default `tmin` / `tmax` and it would make sense. Or you could do `raw.to_epochs(tmin=tmin, tmax=tmax)`.

(Analogously, you can have `epochs.to_raw()` and it would give you a raw object from the concatenated epochs)

kingjr, 2016-02-21:
I'm -1 on epochs.to_raw() . It would create plenty of issue with the subsequent filtering

jasmainak, 2016-02-21:
> I'm -1 on epochs.to_raw() . It would create plenty of issue with the subsequent filtering

Point taken. Probably not such a great idea to go from epochs to raw.

jona-sassenhagen, 2016-02-21:
I'm not sure what the purpose of `.to_raw()` would be, but I like `raw.to_epochs`. It would avoid the `mne.Epochs` or `from mne import Epochs` that's right now just boilerplate. And code wise, it's probably a very small change.

teonlamont, 2016-02-21:
+1 `raw.to_epochs`
-1 `epochs.to_raw`

Eric89GXL, 2016-02-21:
Why add another way to construct epochs from raw with raw.to_epochs instead
of extending support of Epochs(raw)?

jona-sassenhagen, 2016-02-21:
Well I'd think it would be very simple to have raw.to_epochs() just be a call to mne.Epochs(self), wouldn't it?

jasmainak, 2016-02-21:
I feel `raw.to_epochs` is a cleaner approach to the issue @teonlamont raised. It offers exactly what you want to do: quickly look at the epochs data without many intermediate steps / arguments. Just give two arguments (`tmin` and `tmax`) and voila -- you have your epochs object. It's a simple transformation that works in most cases.

To me, `Epochs` is the data container -- which need not necessarily be tied to `raw` (in most cases it does), but can also be read from a file (e.g., EEGLAB set files) or a real-time stream. It is more generic. If you want to give events which don't come from the stim channel, then use this object.

jona-sassenhagen, 2016-02-21:
For the experiments I know, there'd only be very few usage cases for a `raw.to_epochs()` without kwargs.  I just think making `raw.to_epochs()` an alias of `mne.Epochs(raw)` would be a handy shortcut.

kingjr, 2016-02-21:
[I actually got mistaken, I thought we were discussing about `EpochsArray`, which I personally would like to get as much default (`tmin`, `events`, `event_id`) as possible to quickly look up data (from a 3D matrix) without having to `make_info`, define an `events` array etc.)]

Eric89GXL, 2016-02-21:
> To me, Epochs is the data container -- which need not necessarily be tied
to raw (in most cases it does), but can also be read from a file (e.g.,
EEGLAB set files) or a real-time stream. It is more generic. If you want to
give events which don't come from the stim channel, then use this object.

I think of Epochs(raw) as being how you construct epochs from a raw object.
I suspect that is how users think about using that class. There are other
ways to get access to epochs sure -- reading saved ones from disk for
example -- but not from a raw object. In other words, I'm -1 on increasing
the number of ways we will need to answer the question "what function or
method do I use to create epochs from my raw object?", as this proposal
would do, unless there is a compelling reason to.

I don't think events=None case (since I think that's all we're really
talking about) is worth making a new function for. Convenience is slightly
more compelling, but I'm still -0.5 on that (it goes against the "one clear
way" mantra, which I think we violate a bit too often). If other people do
agree on adding the method, though, I don't yet see a reason to make it
anything other than a wrapper to Epochs, with appropriate expansion of
Epochs capabilities.

jasmainak, 2016-02-21:
> I'm -1 on increasing the number of ways we will need to answer the question "what function or method do I use to create epochs from my raw object?", as this proposal would do, unless there is a compelling reason to.

This was exactly the question bothering me when I proposed this. However, I think we have precedence in other packages in the scientific python ecosystem to justify such a move. See for example how you can convert a [Series into a DataFrame](http://pandas.pydata.org/pandas-docs/stable/generated/pandas.Series.to_frame.html) or [a list to a numpy array](http://docs.scipy.org/doc/numpy-1.10.1/reference/generated/numpy.asarray.html). I would consider transforming objects an exception to this maxim.

> I don't yet see a reason to make it anything other than a wrapper to Epochs

I'm fine with that.

jona-sassenhagen, 2016-02-21:
What do you think would be more consistent/MNE style Eric?

I think `epochs = raw.to_epochs()` would be more than convenience; it would also save at least one line of boilerplate in basically every script, demo, everything, while, if at all, cleaning up the namespace.

Eric89GXL, 2016-02-21:
I think using Epochs(raw, ...) has been "the one way to do it" since the
start of the package (something like 5 years?), so there is a lot of
cumulative material and code surrounding it. I'd rather not change all
examples, docs, and teaching materials to use the new method. I actually
find the existing class-instantiation way a bit nicer from an instructional
perspective because it makes it obvious which class users will deal with.
So to me saving a line (which it doesn't do if your import is `import mne`)
isn't worth the disruption to existing patterns. I don't want to end up
like pandas where it feels like the interface, recommended ways of doing
things, etc. seem to change release to release.

jasmainak, 2016-02-21:
> I think using Epochs(raw, ...) has been "the one way to do it" since the start of the package (something like 5 years?), so there is a lot of cumulative material and code surrounding it. I'd rather [...]

https://en.wikipedia.org/wiki/Status_quo_bias ;) On a more serious note, let's wait for other opinions then.

dengemann, 2016-02-22:
No new API methods etc. for this please, YAGNI.
We have Epochs and it did a great job so far. -1 on raw methods, especily if its only for these convenience reasons. It will blow up our APIs and ways to do it and generate lots of additional maintenance work for no clear benefit. 
Let's implement the initial idea in a clean and simple way. Maybe logger.info would be enough.

dengemann, 2016-02-22:
I am really -100 on a raw to epochs method for the reason @Eric89GXL mentioned. For stable APIs you must not change core APIs often. Pandas is a bad example, their API politics are a reason why their package is a PITA to maintain in production code. Eric and I removed pandas entirely for these reasons from some of our projects. Stop "geeking" and put your energy into new computational or viz functionality. 
If you need shortcuts put them in your private library.

jasmainak, 2016-02-22:
I feel that something that has been a pain to users for 5 years doesn't need to continue being a pain to use. Pandas might be a bad example, but even folks at numpy follow the same pattern. It's a common pattern that's in my muscle memory. IMHO, "zen of python" is a good starting point, but there are reasonable use cases where we should allow for exceptions. My 2c.

dengemann, 2016-02-22:
But I a gree with JR that especially the EpochsArray can benefit from sensible defaults.

teonlamont, 2016-02-22:
the `event_id` is completely separate from the method suggestion. I think that since we have internals that handle when None is specified, we should be able to have `Epochs(..., event_id=None,...)` by default.

teonlamont, 2016-02-22:
but to be honest, we should have a function for epochs creation. we are inconsistent with new object construction. for instance, for Raw, we have the io module for the `read_*` functions, for Epochs we construct with the class, for SourceEstimates we use the `apply_inverse` operators, and Time Frequency is completely a mix. there should really be clear nomenclature for making these objects.

dengemann, 2016-02-22:
Friends, listen. Just chatted with @agramfort and he agrees that we should
stop talking about core API changes for a good while. If you don't manage
to see the issue and the costs that go along with this now don't be
worried. In the mean time there are many more exciting and valuable things
to do. For some of you, PhD theses are to be advanced, fascinating work on
viz, signal processing, statistics and code documentation is waiting for
us. And I'm almost 100% sure that the weaknesses that have been existing to
our Epochs API are not actual impediments to all these fascinating things
that are waiting for us. Maybe in a bit you will then understand, that once
you have a working and good API, you don't want to touch it just because
you find another more sexy way to do it. Debian but also numpy and sklearn
are great software projects with a wide user base exactly because they
understood that core API development has to be slow. Think of *really*
slow. Now tripple that. I think we should learn from them and invest our
collective energy accordingly. For us, slow, in the moment this means not
to touch core API issues beyond bugs and documentation. There might be a
time when we see at one glance a new consistent way of doing everything.
But it will feel different, that's not now. These gradual little API
changes lead in to hell directly. A mess is waiting there for us and that's
why we are not going there. Now please re-orient your attention and release
your enthusiasm on other topics that won't drive me and @agramfort mad.
Thank you.

closed 2016-02-22 by agramfort

teonlamont, 2016-02-24:
@dengemann thanks for your opinion but I think that the development of this project often lacks the insights of the typical user. since I've been working on my thesis, I have fallen into the role of a user and I just began noticing things that just sometimes didn't make sense. I have been posting issues to make note but none of them have been for complete overhaul, more to have a document a few of the shortcomings so that when we have the time to collectively review, we can see what works and what doesn't, hence my unfamiliar restraint to just submit PRs. I think we do more or less have a solid api, but some of the things are a bit opaque and inconsistent seeing how it grew from developers at different stages of the repo. we've definitely become more vigilant about maintaining consistency now, although things like having `tmin` and `tmax` in most places then seeing `start` and `stop` in others drives me insane! you then end up facing the issue of perserving for backwards compatibility or break it for uniformity and consistency. I also agree with your sentiments you mentioned elsewhere with the rapidity of adding code and that needs to be balanced with encouraging developers to contribute because if the project stalls than it's sometimes hard to restart it.

tl;dr some inconsistencies bother me and sometimes post issues. Issues are for discussing, they don't necessarily predicate PR and change.

dengemann, 2016-02-24:
If you don't have time to read, to summarize, I'd suggest the following
policy:
- 2+ merge rule for non-core API
- no pull request on core API if a core contributor raises -1 in the
discussion, same for merge
- core API to me is the public constructors of core objects like Epochs,
Raw, Info and designated ways to do things.
- non core API to me is any documentation, viz functions, preprocessing,
statistics, speed and memory improvements, new analyses, new IO.

I think we should agrree on a definition of core-api issues and write a
simple policy. The bottom line should be that for core-api topics we need
maximum consensus and should go slow. It's not ok when discussions emerge,
soon after a PR is issued and merged because silence is tacitly interpreted
as agreement.
Teon I know well that we have these issues but the balance is really
delicate. It is important to discuss them and implement a consistent change
slowly. Many rapid and isolated PRs will just relocate the mess from A to
B. We have many other issues where less consensus is needed. Btw. I'm
getting feedback from PIs that our code is changing too fast, whenever they
have time to continue on an analysis the ways to do it changed again. I am
pro consistencey but more imporatnatly we should not change a running
system, especially not if it means *now*.

It would be very nice if we could decouple core API discussions from our
fast-merging track. As in core python discussion should and can be long and
maybe rejected even after months or a year. I really would feel happier if
we could be sure that whatever we discuss here on core API does not
transform anytime soon into a quickly merged PR. This is the only way how
we can really have stability and controlled development of our interfaces.
As I said repeatedly, in the meantime we have many other things on which we
can advance much faster, with a simple +2 merge rule. In the face of doubt
go for backwards compatibility, we already made it and we gave our
commitment by merging it, people have learned the API and have used it. In
turn, to avoid bad API development, be slow and get it right just from the
beginning. Take time. Don't merge fast on this.

PR #2921 (2016-02-23): [MRG]Changed default values of epoch parameters. Fix issue #2872
Changed `event_id = None`, `tmin = -0.2`, `tmax=0.5`. I changed the value seeing the default in `mne/tests/test_epochs.py`. Reviews and suggestions about the default values please @agramfort @kingjr @teonlamont @jona-sassenhagen @Eric89GXL
closes #2872