mne-tools/mne-python #2873: Layout Enhancement

teonlamont, 2016-02-11:
`find_layout ` doesn't work for at least the KIT system if your object includes non-MEG (auxiliary, refs, etc.). The current solution is to do `pick_types` then call the function, but this seems like something the function should handle.

agramfort, 2016-02-11:
PR welcome :)

teonlamont, 2016-03-05:
@kingjr IIRC you like viz, right? ;)

kingjr, 2016-03-05:
Well... I ll have to fix soon anyway

Eric89GXL, 2016-09-08:
@kingjr is this possible in the next two weeks?

agramfort, 2016-09-15:
FYI we will release in a week. If not done we'll postpone. Just let us know.

kingjr, 2016-09-15:
Not my priority sorry

agramfort, 2016-09-15:
ok removing 0.13 milestone

rodrigohubner, 2017-03-31:
What database is failing in the issue you have opened?
I'm trying to reproduce the problem to find a solution in the code ... What were the steps for the problem with `add_channels` to happen?

Eric89GXL, 2017-03-31:
@rodrigohubner try:

1. use `raw = mne.io.read_raw_kit(..., preload=True)` with some of the data from https://github.com/mne-tools/mne-python/tree/master/mne/io/kit/tests/data
2. use `raw_extra = RawArray(...)` to make a single-channel Raw instance, maybe with MISC type
3. use `raw.add_channels(raw_extra)`
4. use `find_layout(raw.info)` to see if it fails

It's important to get this going soon because the GSoC deadline is approaching. We need to be able to evaluate your coding and coding process...

rodrigohubner, 2017-03-31:
I was able to reproduce this situation and no error occurred.
I believe the error occurred because some of the steps to add the channel were wrong. Here is the code I made with the output:

`import mne
import numpy as np

# open raw kit loaded
raw = mne.io.read_raw_kit('mne/io/kit/tests/data/test.sqd', preload=True)
print "Number of channels before 'add_channels':", len(raw.ch_names)

# create a single-channel raw instance of MISC type
sfreq = 1000 # sampling frequency
times = np.arange(0, 20, 0.01) # use 2000 samples (2s)
sin = np.sin(times * 10)
cos = np.cos(times * 10)
data = np.array([sin, cos])
# definition of channel type and name
ch_types = ['misc', 'misc']
ch_names = ['MISC 033', 'MISC 034']
# create a info object
info = mne.create_info(ch_names=ch_names, sfreq=sfreq, ch_types=ch_types)
# create a raw object with a single channel
raw_extra = mne.io.RawArray(data, info)

# add the misc channels (MISC 033, MISC 034) to raw object
raw.add_channels([raw_extra], force_update_info=True)
print mne.channels.find_layout(raw.info)
print "Number of channels after 'add_channels':", len(raw.ch_names)`

output:

`Number of channels before 'add_channels': 193
<Layout | KIT-157 - Channels: MEG 001, MEG 002, MEG 003 ...>
Number of channels after 'add_channels': 195`

Eric89GXL, 2017-03-31:
Hmm... if that doesn't recreate it, then it's likely that this bug has actually been fixed somehow in the meantime. So I'll close this. @rodrigohubner maybe #4133 is better

teonbrooks, 2017-04-01:
closing because I can't even reproduce what I was complaining about

closed 2017-04-01 by teonbrooks