mne-tools/mne-python #2874: BUG: GAT.plot_diagonal with predict_proba and AUC scoring fails b/c calculating chance levels doesn't work

jona-sassenhagen, 2016-02-11:
```
import mne
from mne.datasets import spm_face
from mne.decoding import GeneralizationAcrossTime

# Preprocess data
data_path = spm_face.data_path()
# Load and filter data, set up epochs
raw_fname = data_path + '/MEG/spm/SPM_CTF_MEG_example_faces%d_3D_raw.fif'

raw = mne.io.Raw(raw_fname % 1, preload=True)  # Take first run

picks = mne.pick_types(raw.info, meg=True, exclude='bads')
raw.filter(1, 45, method='iir')

events = mne.find_events(raw, stim_channel='UPPT001')
event_id = {"faces": 1, "scrambled": 2}
tmin, tmax = -0.1, 0.5

decim = 4
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True,
                    picks=picks, baseline=None, preload=True,
                    reject=dict(mag=1.5e-12), decim=decim, verbose=False)

def auc_scorer(y_true, y_pred):
    from sklearn.metrics import roc_auc_score
    from sklearn.preprocessing import LabelBinarizer
    le = LabelBinarizer()
    y_true = le.fit_transform(y_true)
    return roc_auc_score(y_true, y_pred[:, 1])

# Define decoder. The decision function is employed to use cross-validation
gat = GeneralizationAcrossTime(predict_mode='cross-validation', n_jobs=4,
                               predict_method='predict_proba',
                               scorer=auc_scorer)

# fit and score
gat.fit(epochs)
gat.score(epochs)
gat.plot_diagonal()  # plot decoding across time (correspond to GAT diagonal)
```
gives
`AttributeError: 'function' object has no attribute 'func_name'`

but 
`gat.plot_diagonal(chance=.5)`
works.

I think that counts as a bug? Or should the user-defined scorer be different?

We've discussed this in the context of #2755 , but I think there should be an issue.

Also, maybe we should consider adding a scorer that can inherently deal with `predict_proba`, ie this one (by Jean-Remi).

@kingjr

kingjr, 2016-02-11:
It's a user-specific scorer, so I think it's ok to ask the user to specify chance level. But perhaps a warning would suffice.

kingjr, 2016-02-11:
Also, I think the scorer's chance level should be set in the sklearn API. I'm not aware of this ATM.

jona-sassenhagen, 2016-02-11:
>It's a user-specific scorer, so I think it's ok to ask the user to specify chance level. But perhaps a warning would suffice.

There could be a clearer error message that the problem is in determining the chance level. As-is, the error is rather opaque.

kingjr, 2016-02-11:
agreed.

kingjr, 2016-02-11:
Let's wait until the current beast is merged, so we can do small PR again.

jona-sassenhagen, 2016-02-17:
How about, when #2856 is done, I tackle this one? At the very least by providing an informative error when chance is not a float.

kingjr, 2016-02-17:
Sure! Let's just throw a warning though, and avoid plotting the dashed line.

I'm also wondering whether we shouldn't ask sklearn people whether we should get a `get_chance(scorer, clf)`

kaichogami, 2016-03-04:
@jona-sassenhagen are you working on this? If not, I would like to try this one.

jona-sassenhagen, 2016-03-04:
I'd start there once #2975 is merged, so if you want to start now, it's yours!

kaichogami, 2016-03-04:
@jona-sassenhagen I think you should work on this. I'll work on some other issue! :)

PR #2987 (2016-03-07): MRG safer plotting of bad chance level in plot_diagonal
closes #2874

@kingjr was there any reason for using `scorer.func_name`?
commit a812c102 (2016-03-07): safer plotting of bad chance level in plot_diagonal

closes #2874

@kingjr was there any reason for using `scorer.func_name`?
closed 2016-03-07 by Eric89GXL