mne-tools/mne-python #2875: HED with evokeds

teonlamont, 2016-02-12:
HED tags no longer work as desired if you're using `linear_regression_raw`. the output is a dictionary of evokeds. so for instance you have the tags `aud/primed`, `aud/unprimed`, `vis/primed`, `vis/unprimed` if you wanted to look at the grand average from here of `aud`, you would need use `mne.grand_average` as opposed to `Epochs['aud'].average()`.
I guess the point I'm trying to ask is why is there no container for collection of evokeds. They are usually returned as a list.

dengemann, 2016-02-12:
To avoid API and object flooding. In the past I pulled out the hierarchical
dict part as a function to work with any dictionary. Maybe we can provide
such a tool in `mne.event.py`

jona-sassenhagen, 2016-02-12:
Tthere is an important difference here between averaging and rERPs: you can't really estimate your 2x2 design to give you 0/0, 0/1, 1/0, 1/1 as distinct evokeds, because the design matrix would not be full rank. One of the contrasts is fully redundant - e.g., if you know 0/0, 0/1, 1/0, you can perfectly predict 1/1. This means multicollinearity, an unidentifiable model, and in the end an uninterpretable coefficient series. Instead, you could for example estimate 1. shared by all (just the mean evoked potential), 2. primed/unprimed, 3. auditory/visual (and possibly 4. an interaction). 2 and 3 (and 4) would correspond to difference waveforms.

Viewed like that, HED doesn't make sense for rERPs all that often.

On the other hand, it still does make sense in a few situations, and I think it might be worth having a dict-like object that works with HED selection. However, its `.get()` should IMO always return one evoked object - e.g., return the (unweighted) average of all select effects.

dengemann, 2016-02-12:
-1 on more objects for such a thing. A generic key selector function pulled out of epochs get item would be easier to maintain.

agramfort, 2016-02-13:
> Viewed like that, HED doesn't make sense for rERPs all that often

I agree. You should use an "aud" event in the rERP design to extract the
auditory response.

dengemann, 2016-02-13:
Teon in general, if it helps look at this:

```Python

def subselect_keys(key, mapping, sep='/'):
    if isinstance(key, str):
        key = [key]

    mapping = deepcopy(mapping)

    if any(sep in k_i for k_i in mapping.keys()):
        if any(k_e not in mapping for k_e in key):
            # Select a given key if the requested set of
            # '/'-separated types are a subset of the types in that key

            for k in mapping.keys():
                if not all(set(k_i.split('/')).issubset(k.split('/'))
                           for k_i in key):
                        del mapping[k]

            if len(key) == 0:
                raise KeyError('Attempting selection of keys via '
                               'multiple/partial matching, but no '
                               'event matches all criteria.')
    else:
        raise ValueError('Your keys are bad formatted.')
    return mapping
```

I'd support adding such a thing in some utils module.

jona-sassenhagen, 2016-02-13:
We have the HED code basically twice - in epochy.py in lines 1355ff and 1600ff - once for the epochs get, and once for `equalize_event_counts`. So maybe we should refactor it out and it would be available as a private function anyways.

jona-sassenhagen, 2016-02-13:
Btw a dict of evokeds is a really typical pattern in my MNE code (even when not using rERPs) and I think it's very useful in general, maybe there should be an example somewhere.

dengemann, 2016-02-13:
Yes, exactly I would put such a function somewhere and use it in all other context, it is actually pretty general and works with any dictionary and as you see it also is easy to use custom separators.

jona-sassenhagen, 2016-02-13:
@teonlamont rename into "Refactor HED"? Or make a new issue?

teonlamont, 2016-02-14:
sorry, I was just trying to give a simple example but that isn't the use case where this occurs. The HED is useful when you are trying to account for overlapping events but you are not actually interested in some of the events. you would have the fully factored out HED tag, say it's like w/x/y/z but you are actually only interested in x/y versus x/v.

teonlamont, 2016-02-14:
this could also be repurposed for the refactoring.

jona-sassenhagen, 2016-07-02:
@dengemann if you have the time ...

Eric89GXL, 2016-09-08:
@dengemann do you have time to tackle this in the next two weeks?

Eric89GXL, 2016-09-16:
Removing milestone

PR #3603 (2016-09-26): MRG: Better logic for epoch selection
@jona-sassenhagen do you have time to take a look? This deals with the hierarchical event id stuff and should make it a bit more consistent.

Closes #3599.
Closes #2875.
closed 2016-09-26 by agramfort