mne-tools/mne-python #2878: Warning message in set_channel_types causes KeyError

smerdis, 2016-02-13:
Hi there. I have some data which I read in using this code:

`raw = mne.io.Raw(f, preload=True)`

If I then take a look at an individual channel in the info structure, it looks like this:
```
raw.info['chs'][ch_names.index('EEG 1')]
{'cal': 1.0,
 'ch_name': u'EEG 1',
 'coil_type': 0,
 'coord_frame': 0,
 'kind': 0,
 'loc': array([ 0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.,  0.]),
 'logno': 1,
 'range': 1.0,
 'scanno': 1,
 'unit': 0,
 'unit_mul': 0}
```

Now I want to set the channel type to 'misc' for all channels using set_channel_types(), but it blows up:
```
ch_types = dict([(chnm, 'misc') for chnm in raw.ch_names])
raw.set_channel_types(ch_types)
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-14-fc7277662040> in <module>()
      1 ch_types = dict([(chnm, 'misc') for chnm in raw.ch_names])
----> 2 raw.set_channel_types(ch_types)

/home/despoB/arjun/anaconda2/lib/python2.7/site-packages/mne/channels/channels.pyc in set_channel_types(self, mapping)
    289                 warnings.warn("The unit for Channel %s has changed "
    290                               "from %s to %s." % (ch_name,
--> 291                                                   _unit2human[unit_old],
    292                                                   _unit2human[unit_new]))
    293             self.info['chs'][c_ind]['unit'] = _human2unit[ch_type]

KeyError: 0

> /home/despoB/arjun/anaconda2/lib/python2.7/site-packages/mne/channels/channels.py(291)set_channel_types()
    290                               "from %s to %s." % (ch_name,
--> 291                                                   _unit2human[unit_old],
    292                                                   _unit2human[unit_new]))
```

It blows up because unit_old was 0, and _unit2human doesn't have a key 0. This warning message should check unit_old exists in _unit2human or something.

jasmainak, 2016-02-13:
arff ... we had a similar issue recently: https://github.com/mne-tools/mne-python/issues/2770. Fix welcome.

dgwakeman, 2016-02-13:
Looks like it might be a bug in the reader for that file type. To fix it you could set the unit. If you know for sure what the unit is. To do the change you might need to rescale the units of the data depending on the unit the data are in. We would also welcome a pr fixing the reader to do the unit conversion (if necessary) and setting the right unit type (assuming I am right)

Eric89GXL, 2016-02-22:
> Looks like it might be a bug in the reader for that file type.

This file appears to be a FIF file, so I suspect this file was converted by another package. Is htat right @smerdis? It would be best to fix that converter, but we should also be smarter about dealing with bad conversions.

agramfort, 2016-02-22:
i suspect its fieldtrip

PR #2955 (2016-02-26): ENH : better error message when unit is not known
closes #2878
closes #2879
closed 2016-02-26 by Eric89GXL