mne-tools/mne-python #2884: Adding an option to plot norm values of the grads in plot_evoked_topo?

mmittag, 2016-02-15:
Hi,
It would be great if an option to plot norm values of combined gradiometer pairs (sqrt of (grad12 + grad22)) could be added in plot_evoked_topo. Brain storm does that automatically in its gui (http://neuroimage.usc.edu/brainstorm/Tutorials/TutMindNeuromag) and various groups seem to publish such timecourses of the combined grads.
Much thanks for your help!
Maria

agramfort, 2016-02-15:
can you point to the figure you are referring to?

mmittag, 2016-02-15:
Hi, sorry for the confusion. I did not refer to a figure on the link I provided. The link shows implementation of grad norms with Brainstorm, not the figure I wish to create. I wish to create a figure similar to the one attached, with the difference that norm values of combined gradiometers are illustrated. Plot_evoked_topomap already does a similar norm (RMS feature) by adding `ch_type='grad'`, but I cannot find such element for plot_evoked_topo. thanks!

[image]

agramfort, 2016-02-16:
ok that seems like a legitimate use case and an easy PR.

for a new contributor?

mmittag, 2016-02-16:
That's great. This implementation will be very useful, thank you. It would be an honorable task to contribute, but I have to pass on this one since I'm not that familiar with creating new codes for mne-python.

dengemann, 2016-02-16:
Maybe something for @cjayb ;) all the ingredients needed can be found in plot_topomap which already does the norm remapping once grads are encountered.

cjayb, 2016-02-16:
@dengemann hehe, I'll take a crack at it :)

mmittag, 2016-02-16:
thank you! @cjayb

cjayb, 2016-02-16:
ugh, just took a closer look: not trivial, have to think about layout & possible delayed SSP. Have little time until next week, won't take it personally if someone else solves this in the mean time ;) Certainly a nice to have feature, this.

cjayb, 2016-02-16:
see #2898, help needed

closed 2016-02-23 by agramfort