mne-tools/mne-python #2887: BUG plot_topomap_evoked colormap when specifying axes

cjayb, 2016-02-15:
Consider this (desired behaviour is three topomaps with common vlims and a single colorbar to the right):
```
fig, axes = plt.subplots(1, 4)  # last sp is for the colorbar
first_evo.plot_topomap(..., axes=axes[0], vmin=vmin, vmax=vmax, colorbar=False)
second_evo.plot_topomap(..., axes=axes[0], vmin=vmin, vmax=vmax, colorbar=False)
third_evo.plot_topomap(..., axes=axes[0],vmin=vmin, vmax=vmax, colorbar=True)  # last one with colorbar
```
produces colorbar in third subplot, and nothing in fourth. 

I think I have a very simple fix, seems not to screw up usual behaviour (?)

closed 2016-02-16 by jaeilepp

cjayb, 2016-02-16:
@jaeilepp I think i may have broken `mne-python` :-( I'm getting weird behaviour of `plot_topomap`, can we roll back while I investigate? Alternatively, change it back with a new commit (cleaner?)? 

See this
```
from mne import read_evokeds
from mne.datasets import sample
data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis-ave.fif'
evoked = read_evokeds(fname, baseline=(None, 0), proj=True)[0]
evoked.plot_topomap(times=0.080, ch_type='mag')  # colorbar on topo
evoked.plot_topomap(times=[0.060, 0.080], ch_type='mag')  # colorbar too far to left
```
Very sorry about the hassle...

cjayb, 2016-02-16:
@jaeilepp found the problem in L1240 of `topomap.py`
```
    cpos.x0 = 1 - (.7 + .1 / size) / n_fig_axes
```
Any idea what this is _supposed_ to be doing?

jaeilepp, 2016-02-16:
Hmm. The script you provided seems to be working for me. What is the problem exactly?

jaeilepp, 2016-02-16:
No. I was on a wrong branch. I see what you mean.

cjayb, 2016-02-16:
:) see #2891