mne-tools/mne-python #2897: MAINT/VIZ: replace subtplots with axes divider and tile image output from plot_topomap

dengemann, 2016-02-16:
Our plots don't look as nice as they could in terms of space usage, this gets especially obvious in our joint plot. I'd suggest to remove the less precise repeated calls to plt.subplots. We could instead use the divider pattern, e.g. see here http://martinos.org/mne/stable/auto_tutorials/plot_spatio_temporal_cluster_stats_sensor.html)
And pass an entire properly masked tiled numpy array to that axes.
Of course, some exploration and tuning is needed here.

# viz_people @jaeilepp @jona-sassenhagen @cjayb ...

cjayb, 2016-02-16:
> And pass an entire properly masked tiled numpy array to that axes.

+1, for sure! But what do you mean with this?

dengemann, 2016-02-16:
Well the devider pattern is easier to use if you use one axis.
So you make one long image from a numpy array and fill the axis with this instead of making new subplots in guessing-dynamic way. Ok?

dengemann, 2016-02-16:
I just remember that it was not so trivial to get the masking with the topos right ... sounds terrible but probably it would be easier to fake file object, write the topo as it is and read it as image and scale it into the desired slot ...
Anyways, little viz challenge :)

Eric89GXL, 2016-02-26:
This will also likely make the plotting faster by reducing the number of axes and draw calls

dengemann, 2016-02-26:
> This will also likely make the plotting faster by reducing the number of axes and draw calls

+1

jona-sassenhagen, 2016-04-07:
I agree this would be great to have, plotting is too slow. But looking over the code, I don't think I could do it ...

agramfort, 2017-07-22:
closing as non-specific enough.

closed 2017-07-22 by agramfort