mne-tools/mne-python #2899: BUG: pickle error with ancient evoked

kingjr, 2016-02-17:
@dengemann 
I'm not sure whether this has been reported elsewhere, but in case:

I had saved `evoked` objects together with some other data with `pickle` a few months ago. eg.
```
with open('foo.pkl', as f):
    pickle.dump([evoked, bar, joe], f)
```

Now, if I try to `pickle.load` it, it crashes:

```
/home/jrking/anaconda/lib/python2.7/pickle.pyc in load(file)
   1382 
   1383 def load(file):
-> 1384     return Unpickler(file).load()
   1385 
   1386 def loads(str):

/home/jrking/anaconda/lib/python2.7/pickle.pyc in load(self)
    862             while 1:
    863                 key = read(1)
--> 864                 dispatch[key](self)
    865         except _Stop, stopinst:
    866             return stopinst.value

/home/jrking/anaconda/lib/python2.7/pickle.pyc in load_reduce(self)
   1137         args = stack.pop()
   1138         func = stack[-1]
-> 1139         value = func(*args)
   1140         stack[-1] = value
   1141     dispatch[REDUCE] = load_reduce

/home/jrking/anaconda/lib/python2.7/copy_reg.pyc in _reconstructor(cls, base, state)
     48         obj = object.__new__(cls)
     49     else:
---> 50         obj = base.__new__(cls, state)
     51         if base.__init__ != object.__init__:
     52             base.__init__(obj, state)

TypeError: dict.__new__(Info): Info is not a subtype of dict
```

This works fine when checking a previous version:e.g. `4b89a0472f955e369afac89df1f61bca4bc34157`

I guess some may advice not to use pickle for this kind of saving... Which is true, but this issue nevertheless remains a problem..

dengemann, 2016-02-17:
Welcome to pickling hell. This is why we have fd5 and fiff, with pickling you can't ensure that these things won't 
happen and you can find many materials when googling for this. We therefore discourage its use (unless you know what you do and have to do). We cannot give any support for these kind of things unfortunately. Pickling is not API-safe, it is only useful for temporary dumping as used by joblib. Never use it to arcive data, you are bound to the runtime and versions of pickling time. 
cc @cmoutard I told you too ;)

closed 2016-02-17 by agramfort

kingjr, 2016-02-24:
Mmh, I predict that the class change of Info will annoy quite a few people though...