mne-tools/mne-python #2904: Interactive joint_plot

teonlamont, 2016-02-17:
Is it possible to interactive reselect the time points in the joint_plot?

If not, that would be a great feature to have.

jona-sassenhagen, 2016-02-17:
I couldn't do it - maybe @jaeilepp can check if it's even possible.

kingjr, 2016-02-17:
You should be able with http://matplotlib.org/users/event_handling.html.

However it won't be easy to change the window length of the selected ERF/P.

jona-sassenhagen, 2016-02-17:
I don't know anything about interactive plotting, but the problem I see is that plot_joint is basically a wrapper for `evoked.plot_topomap` and `evoked.plot`. So you'd have to propagate it somehow. Anyway, I can promise I won't even attempt doing this :(

dengemann, 2016-02-17:
improving the behaviour of joint_plot would be more important in the first
place, we need to revamp the interior design at some point to correctly
fill spaces of topos an the canvas. and for interactive you can still just
plot one buttefly and one topomap.

jona-sassenhagen, 2016-02-17:
That would be my intuition, too - plot_joint is kind of an offline version of something you could already do online.

teonlamont, 2016-02-18:
this could be bundled in the revamp. the interactive nature could be shared the way the evoked plot already handles this.