mne-tools/mne-python #2909: block parameter in epoch_plot doesn't work as intended

ualsbombe, 2016-02-18:
Dear Developers

The block argument in plot_epochs doesn't work as intended. It keeps evaluating the code, and thus, in this particular code (see below), the saved output is the same as the input. I am using mne version 0.11.0 and running from Spyder in an Anaconda-environment.

Best

Lau

`#==============================================================================
# MANUALLY CLEAN EPOCHS
#==============================================================================
epochses = []

for i in range(len(positions)):
    input_file = './3_epochs/' + positions[i] + '_eog_auto_cleaned-epo.fif'
    epochs = mne.read_epochs(input_file)
    mne.epochs.plot_epochs(epochs, picks=[0], block=True) ## only MISC010
    epochses.append(epochs)
    output_file = './4_cleaned_epochs/' + positions[i] + \
                '_manually_cleaned-epo.fif'
    epochs.save(output_file) `

jaeilepp, 2016-02-18:
This seems to work for me. What is the issue actually? How do you expect the saved epoch to differ?

agramfort, 2016-02-18:
I suspect its a matter of marking bad epochs visually

I suspect a spyder issue

ualsbombe, 2016-02-18:
Sorry, that wasn't explicit. I want to be able to mark bad epochs and drop them before it saves it (agramfort's suspicion is thus right)

jaeilepp, 2016-02-18:
I tried running it in spyder, but I still can't reproduce the problem. Could you try turning interactive mode on by
```
import matplotlib.pyplot as plt
plt.ion()
```
and see if it works for you. It works with both modes for me though.

ualsbombe, 2016-02-18:
Tried both plt.ion() and plt.ioff(). I still get the same kind of behaviour. I'm running it in ipython, which seems to be making the difference. When running it from a normal python console, it works as intended, I just realized.

jasmainak, 2016-02-18:
can you tell us what is the version of matplotlib you are using?

```py
import matplotlib
matplotlib.__version__
```

ualsbombe, 2016-02-18:
It's 1.5.1

ualsbombe, 2016-02-18:
All relevant versions:
Anaconda: 2.5.0
Python: 2.7.11
IPython: 4.0.3
Spyder: 2.3.8
mne: 0.11.0
matplotlib: 1.5.1

agramfort, 2016-02-18:
it's a config thing in spyder.

it's unrelated with mne

ualsbombe, 2016-02-18:
Ok, that's comforting. Do you where exatcly in the config?

agramfort, 2016-02-18:
sorry I don't use spyder

Eric89GXL, 2016-02-18:
It might be worth asking over on the Spyder repo issues to see what they say. FWIW I use Spyder but always use Python interpreters instead of IPython ones, and ioff / block is respected there.

ualsbombe, 2016-02-19:
Ok, will do that!
Thanks!

closed 2016-02-19 by ualsbombe