mne-tools/mne-python #2919: seaborn

kingjr, 2016-02-19:
If one is up for it, it would be great to get the viz compatible with [seaborn](https://github.com/mwaskom/seaborn) as it is become more and more standard for scientific plotting.

At the moment, we still get some weird viz every now and then: e.g. the evoked plot lines are behind the grid.

[image]

What do you think?

Eric89GXL, 2016-02-19:
Sounds good. Probably just need to uniformly bump up our z values

jona-sassenhagen, 2016-02-20:
I like seaborn, I think it's a great lib. I'd be willing to contribute to this.

But note seaborn is basically a one-man job, and thus it's sometimes not always as responsive as one could wish for.

dengemann, 2016-02-20:
Apart from what Jona says,
I think we probably agree that wed don't want to use Seaborn as default or
use it in our examples. It's very easy anyways to use it yourself and we
should not add more dependencies and I really prefer to get viz right at a
diagnostic level. We don't want beautiful but functional plots that really
confront you with the reality of your data instead of making you dream. You
can easily leverage the existing Python eco-system to make beautiful plots
for your papers.
However I agree that we should make it as easy as possible to use 3rd party
viz libraries such as Seaborn, ggplot, mpld3, bokeh, plotly et al.
If this means to adjust a few defaults I'm fine with this. But it should
not interfere with standard "ugly" default matplotlib-style diagnostic
plotting.
This also means that we should not optimize things for Seaborn only, there
are many potentially interesting 3rd party libraries.

jasmainak, 2016-02-20:
looks like an easy fix for a new contributor. I'll mark this issue (bumping up the zorder) as easy and change the title if @kingjr is okay with it. I agree with Denis that our plots are more meant for diagnostics rather than "publication-ready".

teonlamont, 2016-02-20:
isn't someone working on `mne.io.raw_to_publication`? that pr is very welcome ;)

Eric89GXL, 2016-02-21:
+1 but it has to work for all input file formats, experimental paradigms,
and destination journals

jona-sassenhagen, 2016-02-21:
`mne.io.raw_to_publication(fname, journal='Nature')`

kingjr, 2016-02-21:
> mne.io.raw_to_publication(fname, journal='Nature')

`journal="Nature"` should be default, shouldn't it?

dengemann, 2016-02-22:
But don't add it to the raw objet, ok?

PR #2957 (2016-02-26): MRG+2: Better zorder
Closes #2919.

Ready for review/merge from my end. @jaeilepp do you have time to look?

Running `plot_from_raw_to_epochs_to_evoked` with `import seaborn` added to the top, before/after:

[image]

[image]

 [image]

[image]
closed 2016-02-26 by Eric89GXL