mne-tools/mne-python #2922: FIX: warning_once for raw edf with interpolated chans

kingjr, 2016-02-22:
@Eric89GXL https://github.com/mne-tools/mne-python/commit/e0cec488bd67451ccfa8c14d9267353472f109ef introduces a warning, which can be overwhelming
[image]

I guess we'd need a warning_once in this case.

I actually don't understand what going on, nor why this interpolation is needed.

Eric89GXL, 2016-02-22:
The warning should be emitted elsewhere in the code (outside of whatever looping is being done). Do you have time to look into it?

kingjr, 2016-02-22:
> The warning should be emitted elsewhere in the code (outside of whatever looping is being done). 

What do you mean?

Eric89GXL, 2016-02-22:
The warning is emitted somewhere inside a loop, as evidenced by how many times it gets printed. It should instead only be emitted once, preferably when the file is initially read from disk with `read_raw_edf`, at a level *outside* the loop that it's currently in.

closed 2016-03-16 by Eric89GXL