mne-tools/mne-python #2927: Percision issue

kambysese, 2016-02-22:
@Eric89GXL How is it possible to have HP time points less than the first sample offset in the data, i.e., can it be caused by timing/sequence of events the operator initiates a recording?
I am getting a value error for datasets e.g., [where condition]( https://github.com/mne-tools/mne-python/blob/master/mne/preprocessing/maxwell.py#L688) is not satisfied.

`raw.first_samp/raw.info['sfreq'] = 43.333333333333336`
`t = pos[:, 0] = 43.333`
`(t < t_off).any()` or  43.3333 < 43.333333333333336 evaluates true.  Is there not a more fail safe float comparison available for such a case? WDYT of `assert_allclose` or <=?

> Traceback (most recent call last):
>   File "/home/ktavabi/Projects/ilabs/analysis_scripts/kubi/clinical.py", line 107, in <module>
>     print_status=True
>   File "/home/ktavabi/Projects/mnefun/mnefun/_mnefun.py", line 558, in do_processing
>     outs[ii] = func(p, subjects, run_indices)
>   File "/home/ktavabi/Projects/mnefun/mnefun/_mnefun.py", line 727, in run_sss
>     run_sss_localy(p, subjects, run_indices)
>   File "/home/ktavabi/Projects/mnefun/mnefun/_mnefun.py", line 982, in run_sss_localy
>     head_pos=pos, regularize=reg)
>   File "<string>", line 2, in maxwell_filter
>   File "/home/ktavabi/Projects/mne-python/mne/utils.py", line 596, in verbose
>     return function(*args, **kwargs)
>   File "/home/ktavabi/Projects/mne-python/mne/preprocessing/maxwell.py", line 249, in maxwell_filter
>     head_pos = _check_pos(head_pos, head_frame, raw, st_fixed)
>   File "/home/ktavabi/Projects/mne-python/mne/preprocessing/maxwell.py", line 660, in _check_pos
>     % (t[0], t_off))
> ValueError: Head position time points must be greater than first sample offset, but found 43.3330 < 43.3333

Eric89GXL, 2016-02-22:
Yeah we should make our check more flexible, probably with `_time_mask`

Eric89GXL, 2016-02-26:
fixed in recent PR

closed 2016-02-26 by Eric89GXL

kambysese, 2016-02-26:
Thank you sir

kambysese, 2016-03-02:
@Eric89GXL what is `_time_mask` doing? FYI still hitting this for a case where chpi T0 is 22 and first sample offset is 23

Eric89GXL, 2016-03-02:
It's trying to intelligently pick time samples. You shouldn't have a t0 of 22 if you `raw.first_samp / raw.info['sfreq'] == 22`. Can you upload the files somewhere for me to look at?

kambysese, 2016-03-02:
To get around this I've been dropping the first line of the pos files, so this particular case is lost for now. I do however I have numerous datasets (e.g., [here](https://www.dropbox.com/sh/fu29ap9p2120l39/AACHWn11-Kci4Qz0H1JBr-mLa?dl=0)) throwing this exception for much smaller discrepancies.

Eric89GXL, 2016-03-02:
If you send me the path to the raw file I can easily recalculate the pos
using MF

Eric89GXL, 2016-03-02:
(It would be useful to work in the worst offender)

kambysese, 2016-03-02:
Sorry, I don't remember who the worst offender was. I've been painfully trying to make it through 66 datasets and between this error and the `scipy.linalg` svd error it's been a nightmare. Every third dataset has thrown one of these errors.

kambysese, 2016-03-03:
Uploaded two raw files to 

> kasga::/data01/kam/sandbox

If I find the worst offender I'll let you know.

kambysese, 2016-03-05:
@Eric89GXL were you able to access the files on Kasga?