mne-tools/mne-python #2928: BUG: _time_mask does not work properly for non-uniformly spaced inputs

Eric89GXL, 2016-02-22:
See:
```Python
>>> from mne import _time_mask
>>> import numpy as np
>>> _time_mask(np.array([0.4]), tmin=1., sfreq=1000.)
array([ True], dtype=bool)
>>> _time_mask(np.array([0.4, 1]), tmin=1., sfreq=1000.)
array([False,  True], dtype=bool)
>>> _time_mask(np.array([0.4, 1]), tmin=0.5, sfreq=1000.)
array([ True,  True], dtype=bool)
```
@jaeilepp I think you wrote the `deltas.min()` stuff that causes this, can you fix it?

Eric89GXL, 2016-02-22:
(in case it's not clear, the first and third cases are bugs, but the middle one is fine)

jaeilepp, 2016-02-23:
I'm not sure why the third case is a bug. I thought we wanted the ``_time_mask`` to be inclusive i.e. 0.5 is closer to 0.4 than 1. Unless you mean that the 'closeness' should be defined with sfreq taken into account.

PR #2930 (2016-02-23): MRG: Fix time mask
@jaeilepp see if you agree with the fix.

The critical thing I think is that we should only try to fudge the limits if `sfreq` is provided. If it is not, then we have no way of knowing how far apart from `tmin` and `tmax` are from the true original time values. Thinking about what to do with e.g. `_time_mask(np.array([1.], tmin=1.001, sfreq=None)` should hopefully make this clear. If in the original data had `sfreq=100.`, then the extra `0.001` is probably numerical error, and the point should be included, if `sfreq=10000.` then it probably isn't, and the point should be exc [... cut]
closed 2016-02-23 by jaeilepp