mne-tools/mne-python #2931: off-by-1 error in notch filter (multitaper)

lennyvarghese, 2016-02-22:
I've been running into this on occasion when using the multi-taper line noise removal:
ValueError: operands could not be broadcast together with shapes (1,1,245411) (8,245412) 

I'm attaching a log file with the error. Note that this doesn't always happen, but I haven't been able to figure out the specific circumstances that trigger the error.

I've managed to get around this on a temporary basis by commenting out 1142-1143 in filter.py, but that winds up taking quite a bit longer for long pieces of data...

Any ideas?

Log file:
[preprocess_eeg_data.txt](https://github.com/mne-tools/mne-python/files/141563/preprocess_eeg_data.txt)

lennyvarghese, 2016-02-22:
(I'm guessing this is one for @Eric89GXL)

agramfort, 2016-02-23:
can you try to provide a script to replicate? eventually using a fake
RawArray of length 1,1,245411 or 1,1,245412?

thx

Eric89GXL, 2016-02-23:
Replicated with:
```
import numpy as np
import mne
data = np.empty((1, 245411))
raw = mne.io.RawArray(data, mne.create_info(1, 1000., 'eeg'))
raw.notch_filter([60.], method='spectrum_fit')
```
working on it now

Eric89GXL, 2016-02-23:
bug in upstream `nitime` code, will fix here and submit a patch to them

PR #2934 (2016-02-23): MRG: Fix multitaper
Closes #2931.

Upstream bug https://github.com/nipy/nitime/pull/140.
closed 2016-02-23 by agramfort