mne-tools/mne-python #2933: ENH: speed up ICA application and support partial weight matrices

dengemann, 2016-02-23:
We could probably improve our ICA application speed by following the fieldtrip model of assembling a new projection matrix from selected components and subtracting that from eye(n_channels).
A second benefit would be that we could deal with partial mixing and unmixing matrices (only giving you a subset of components).

cc @agramfort @jona-sassenhagen

dengemann, 2016-02-23:
Tested a bit with some dirty code and it looks rather promising.

agramfort, 2016-02-23:
oups...

Eric89GXL, 2017-03-21:
@dengemann maybe this could be a target for the sprint?

dengemann, 2017-03-21:
Yes, maybe.

larsoner, 2021-02-19:
I think we probably don't need this since it doesn't seem to be a processing bottleneck, and we recently refactored the component selection code

closed 2021-02-19 by larsoner