mne-tools/mne-python #2935: Reading Xfit dipole file

kambysese, 2016-02-23:
@Eric89GXL attaching xfit .fif in compressed file containing single ecd waveform and goodness of fit time series. You can also find [here] (https://www.dropbox.com/s/lurthfo1hy9tzvg/fam_115_LH.fif?dl=0) if attachment doesn't work. 
[fam_115_LH.zip](https://github.com/mne-tools/mne-python/files/143288/fam_115_LH.zip)

Like I said `read_meas_info(fid, tree, clean_bads=True)` seems to get at `meas` but not `info` b/c there is no isotrak, unable to make much sense out of the TAG structure.

Eric89GXL, 2016-02-24:
@kambysese okay with you if I add this to the mne-testing-data repo publically available?

kambysese, 2016-02-24:
> @kambysese okay with you if I add this > to the mne-testing-data repo
publically > available?

Fine by me.

Eric89GXL, 2016-02-24:
Great, thanks. PR made: #2935

Eric89GXL, 2016-02-24:
Whoops I mean #2942

PR #2942 (2016-02-28): MRG: Allow reading and plotting of xfit dipoles
Closes #2935.

Ready for review/merge from my end.
closed 2016-02-28 by agramfort