mne-tools/mne-python #2937: test CTF pipeline using the phantom data

agramfort, 2016-02-24:
I've just been told that we may have bugs in refs channels in our CTF code (present in matlab too apparrently).

to validate our code we should use the phantom data that BST is also using:

http://neuroimage.usc.edu/brainstorm/Tutorials/Phantom

cc @ftadel

Eric89GXL, 2016-02-24:
That would explain why the bst tutorial never looked right

Eric89GXL, 2016-02-24:
@agramfort I don't see a reference for where the dipole actually gets localized. Do you have Brainstorm? Can easily you try running the suggested pipeline and report the results? If not I can give it a shot.

agramfort, 2016-02-24:
ping @ftadel

ftadel, 2016-02-24:
I'm currently fixing the issue and writing code to display correctly the references, so we can have a nice visual reference of what the CTF reference coils look like. It may take me a few more days before I fix everything down to the dipole fitting and update the tutorials. I'll let you know when I'm done.

Eric89GXL, 2016-02-24:
Thanks for the quick response @ftadel, looking forward to the updates.

Eric89GXL, 2016-03-16:
Based on the website it looks like this might be done now

ftadel, 2016-03-16:
Yes, the CTF part is finished and properly documented. It includes the equivalent .fif equivalent to the .ds folders produced with the last update of Matti's converter.

We're waiting for equivalent recordings from the Cleveland Clinic (Neuromag) and hopefully from various 4D sites. I will update the sample_phantom.zip package, the scripts and the website as we get these.

Eric89GXL, 2016-03-16:
Great, thanks for the info @ftadel

agramfort, 2016-03-16:
@Eric89GXL are you on it?

Eric89GXL, 2016-03-16:
I'll see if I can get to it in the next few days

Eric89GXL, 2016-03-21:
I don't think I'll have time for this before 0.12, @agramfort okay with you to move it to 0.13?

Eric89GXL, 2016-03-21:
We could/should probably tackle it at the same time as the bst_auditory stuff

agramfort, 2016-03-21:
fine with me.

no time to help either

PR #3407 (2016-07-11): MRG: phantom tutorials
Tutorials for the [Elekta] (http://neuroimage.usc.edu/brainstorm/Tutorials/PhantomElekta) and [CTF] (http://neuroimage.usc.edu/brainstorm/Tutorials/PhantomCtf) phantoms from the Brainstorm datasets. Our localization is similar for CTF, and slightly better for Elekta because I applied `maxwell_filter` to it (and similar otherwise).

Closes #2937.
closed 2016-07-11 by Eric89GXL