mne-tools/mne-python #2944: Q: Get chan types?

kingjr, 2016-02-25:
Since we can't pickle evoked data (https://github.com/mne-tools/mne-python/issues/2899, again today, big PITA to read old data), and it is sometimes not reasonable to separately store MEG data from other variables (e.g. stats, conditions etc), I was thinking of keeping the necessary MEG `data` and `info` to recreate an `EvokedArray` for later plotting.

I think most params are easy to get (`ch_names`, `tmin`, `sfreq`) but `ch_types`, which, from what I've seen, needs to be created manually (there isn't an `evoked.info['ch_types']`).

If I'm not mistaken, could we add such functionality, or this would create some API issues? Or better, is there an easier way to approach this kind of issue?

Eric89GXL, 2016-02-25:
We already have read_meas_info, if we don't have write_meas_info it would
be easy to create if you can live with separate files and tar them or
something. But meas info plus data is pretty much what an evoked data file
contains, so I'm guessing you want something integrated.

write_hdf5 can also write info dicts (plus a lot of other Python objects),
or at least a modified version with the recent changes. Best of all hdf5 is
a standard format, so should be future compatible.

Eric89GXL, 2016-02-25:
Also, keep in mind the meas info in general contains much more information
than just channel type, especially for MEG data.

dengemann, 2016-02-25:
Yes we have IO for infos in fiff and they work also with hdf5. I very often store just one info next to an array. So I can do what you described for plotting. But please abandon pickles, they will work against you at some point in my experience. Another question: `write_eovkeds` does not do the job? For getting the channel types you have can use the API variant

`mne.io.pick.channel_type`

which will look up `info['chs'][i]['chan_type']`

you can then loop over channel indices to get what you're looking for.

dengemann, 2016-02-25:
FYI: http://www.benfrederickson.com/dont-pickle-your-data/

dengemann, 2016-02-25:
FYI ctd.: http://stackoverflow.com/questions/21752259/python-why-pickle

bottom line: pickling is a good idea for implementing joblib and related things.

kingjr, 2016-02-25:
Ok, I'll avoid pickle from now on and give a test to hdf5. Adding example in the cookbook is definitely a good idea https://github.com/mne-tools/mne-python/issues/2945.

dengemann, 2016-02-25:
take a look at https://github.com/h5io/h5io we're maintaining a copy in mne.externals

PR #3024 (2016-03-17): MRG: FAQ update
A bunch of touch-ups and additions to the FAQ to get us closer to 0.12.

I also updated our `conf.py` to look like the upstream one from `sphinx-gallery` because otherwise I couldn't build the doc :(

Closes #3021.
Closes #3009.
Closes #3002.
Closes #2945.
Closes #2944.
closed 2016-03-17 by dengemann