mne-tools/mne-python #2953: ENH: Speed up topo plots

Eric89GXL, 2016-02-26:
So far only `_plot_evoked_topo` has been sped up, but anything else using `mne.viz.topo._plot_topo` could benefit from the same treatment. That means `mne.time_frequency.tfr.AverageTFR.plot_topo` and `mne.viz.topo.plot_topo_image_epochs`.

The transition involves taking what was formerly e.g. 306 individual axes and making it one single axes object, where data are plotted with offsets to be in the correct locations. Take a look at how `_plot_timeseries` and `_plot_timeseries_unified` work to see the types of changes that need to be made.

Now that we are unifying plotting in this way, `iter_topography` actually is not a great way to do the plotting. For the unified methods, plotting them directly in a for loop would allow construction of `Collections`, which should further speed up plotting. It would also make the code a little bit easier to follow I think.

jasmainak, 2016-02-26:
This would be great to have! The plotting functions are pretty slow

Eric89GXL, 2016-02-26:
If you have any particular slow plotting functions identified, if you could paste the timing of the draw call, with an example or a few lines to reproduce it, it would help the next time one of us gets bored and wants to work on it.

closed 2016-03-23 by agramfort