mne-tools/mne-python #2960: BUG: Raw.add_event and crop

kingjr, 2016-02-26:
Adding an event to `raw` and cropping `raw` afterwards doesn't move the event time.

```
from mne.io import RawArray
from mne import create_info
info = create_info(ch_names=['stim'], ch_types=['stim'], sfreq=1)
raw = RawArray([[0] * 1000], info)
events = [[200, 0, 1]]
raw.add_events(events, 'stim')
print(raw._data[0, 200])  # returns 1, as expected
raw.crop(100, None)
print(raw._data[0, 200])  # returns 1, not as expected
print(raw._data[0, 100])
```

Note that `raw.first_samp` is always 0.

cc @Eric89GXL

teonlamont, 2016-02-26:
just looked at the signature. in raw.crop(), copy=True by default

kingjr, 2016-02-26:
My bad it's was a copy issue

kingjr, 2016-02-26:
@teonlamont yes!

closed 2016-02-26 by kingjr