mne-tools/mne-python #2965: Checks to make sure that sfreq is a float

choldgraf, 2016-02-29:
Hey all - a labmate of mine was running into some weird problems because he had input a sampling frequency that was an integer, and the function was assuming that sfreq was a floating point number.

In this case the problem came up when trying to create an EpochsArray object using an info object that had an integer as the sfreq instead of a float. It was trying to infer a tmax, but the integer division was messing up the calculation.

I saw that `create_info` and most other reading functions auto-convert to a float, but there seem to be some instances where this doesn't happen. E.g.:

```
epochs.resample(10.)
type(epochs.info['sfreq'])
```
== float

and 

```
epochs.resample(10)
type(epochs.info['sfreq'])
```
== int

So I wonder if this is a problem that might pop up elsewhere, in which case is there a more robust way to ensure that sampling frequency is a float?

Eric89GXL, 2016-02-29:
Did they use `create_info` to make the `Info` object, or some other (potentially unsupported) method? If it was `create_info` it should cast to `float` assuming they're on up-to-date master:

https://github.com/mne-tools/mne-python/blob/master/mne/io/meas_info.py#L1522

choldgraf, 2016-02-29:
I asked them the same thing - I think that they manually changed the sampling frequency because they thought that another function wasn't changing it for them. But per the example above you could have created an info object normally and then accidentally changed the sampling frequency to an integer with `epochs.resample`.

https://github.com/mne-tools/mne-python/blob/master/mne/epochs.py#L1482

I guess my question is whether this is just a matter of fixing L1482 to force a float, or if it could be a more general issue worth looking into.

Eric89GXL, 2016-02-29:
That is indeed a bug with `epochs.resample`, it should cast to float. If we ever get around to solving #2860 it will make it so users can't so easily shoot themselves in the foot by modifying `Info` themselves like that.

agramfort, 2016-02-29:
fixed by 5df659b6e2454a835ec6b054170467abfc0de13b

closed 2016-02-29 by agramfort

choldgraf, 2016-02-29:
you beat me to it!