mne-tools/mne-python #2981: BUG: bst_* datasets broken (and thus CircleCI)

Eric89GXL, 2016-03-03:
CircleCI is failing because the copy.com downloads are failing. We need to:

1. Update our URLs to use `https://mne-tools.s3.amazonaws.com` (you can see the paths in there) instead of `copy.com`.
2. I think I also changed the archives to be `gz` instead of `bz2` files (much faster decompression with only a little bit bigger files), but just changing the filename should make this work as type is autodetected.
3. For at least bst_auditory I made it use CTF data instead of the FIFF-converted versions, but hopefully this will only require minimal changes in the example scripts to get working.

@jaeilepp do you have time to try it? If not, I'll have time later in the week.

jaeilepp, 2016-03-03:
Yeah I can try.

Eric89GXL, 2016-03-03:
By the way in circle I think the only bst dataset you need to run examples
might be raw, not auditory or the other one. Worth checking while you're in
there.

closed 2016-03-04 by agramfort