mne-tools/mne-python #2983: Q? objects accessible from mne

kingjr, 2016-03-03:
I regularly wonder why some objects/function are loadable at the root of `mne` whereas some are not. Is there a rule behind such choice?

* The classic example is `Raw`, only accessible from `mne.io` whereas `Epochs` and `Evoked` are accessibles from ` mne`
* Today's example for me was `concatenate_epochs`, which, unlike `concatenate_events` and  `concatenate_raws` isn't accessible from `mne`.

teonlamont, 2016-03-04:
the motivation is that `mne.io` is a grouping for all the raw object constructors, etc. I think the general principle is to have most these accessible from the root. maybe those functions were added later and weren't included in the `__init__`

agramfort, 2016-03-04:
indeed mne.io is for raw files.

historically it contained the fiff_* functions from matlab are the
mne.* was for mne_* functions.

The concatenate_epochs is not in mne.* but I agree it should be. PR welcome.

dengemann, 2016-03-04:
Another reason for this pattern is that we don't always expose functions in the main namespace in the beginning. When we added `concatenate_epochs` it was not clear who beyond me and a few other persons would benefit from it. So the rule is to avoid API flooding, the main name space should be for the most common and important functions. Specialized functions for remain in the module of their domain, e.g., see all the tools we have in events.
I would not be opposed to also exposing the concatenate epochs function there, this is probably indeed a common use case. Simply import in in the main init.

kingjr, 2016-03-04:
So 3 arguments: 1) common API objects get the root shortcut, 2) minimize flooding the root module and 3) historical reasons. I guess I don't have anything much cleaner to suggest... so I'll close the issue.

Regarding to `concatenate_epochs`, this should be pretty common, since most people record their experiment in 5-10 minutes chunks. I'll do a PR

closed 2016-03-04 by kingjr

dengemann, 2016-03-04:
> I'll do a PR

+1