mne-tools/mne-python #2986: Problem saving dates

gcwarner1, 2016-03-04:
I'm trying to change the raw.info['subject_info']['birthday'] field in some files and am unable to do so. Here is the code I'm using.

`raw = Raw(f, preload=False, verbose=False)
raw.info['subject_info']['birthday'] = (01, 01, 1900)
raw.save('new_fif_raw.fif)`

when I open up new_fif_raw.fif to check if it worked I still see the original birthday. I change other fields and they update successfully. What is going on?

agramfort, 2016-03-04:
birthday is not a standard key in the measurement info. We don't write all
the keys to disk but just the standard ones.

Eric89GXL, 2016-03-04:
> raw.info['subject_info']['birthday'] = (01, 01, 1900)

This line has a `SyntaxError` in it -- `01` isn't a valid integer, you probably want `1`. Perhaps you are hitting this error and not noticing (running the `.save()` line anyway)?

Also, note that the order is Y/M/D not M/D/Y, so you'd want to do:

```
raw.info['subject_info']['birthday'] = (1900, 1, 1)
```
Verifying:
```
>>> import os.path as op
>>> import mne
>>> raw = mne.io.Raw(op.dirname(mne.__file__, 'io', 'tests', 'data', 'test_chpi_raw_sss.fif')
>>> raw.info['subject_info']['birthday']
(1983, 12, 19)
>>> raw.info['subject_info']['birthday'] = (1900, 1, 1)
>>> raw.save('test_raw.fif')
>>> mne.io.Raw('test_raw.fif').info['subject_info']['birthday']
(1900, 1, 1)
```
Can you confirm?

Eric89GXL, 2016-03-04:
@agramfort it actually does live in `subject_info` :)

agramfort, 2016-03-04:
>
> @agramfort <https://github.com/agramfort> it actually does live in
> subject_info :)
>

oups...

dgwakeman, 2016-03-07:
@gcwarner1, am I right in thinking you noticed this field doesn't get changed in the anonymize function? Do you have code that fixes that?

Eric89GXL, 2016-03-07:
@dgwakeman no, it should be deleted (in both MNE-C and mne-python) because the entire `info['subject_info']`, which ends up being written as its own block in the FIF file, gets deleted, see:

https://github.com/mne-tools/mne-python/blob/master/mne/io/meas_info.py#L373

closed 2016-03-07 by Eric89GXL

gcwarner1, 2016-03-07:
Looks like the anonymize function works for this. However, it does not edit or delete raw.info['meas_date']. Shouldn't that be included in the anonymization function as it is technically PHI?

kingjr, 2016-03-07:
> Looks like the anonymize function works for this. However, it does not edit or delete raw.info['meas_date']. Shouldn't that be included in the anonymization function as it is technically PHI?

I'm -.5 on this, the rec date isn't an individual feature, and is critical when there's a mixup in subjects filenames.

dgwakeman, 2016-03-07:
although it definitely qualifies as PHI see #3:
http://cphs.berkeley.edu/hipaa/hipaa18.html

kingjr, 2016-03-07:
Fair enough

Eric89GXL, 2016-03-07:
@mshamalainen do we need to add the measurement date to `mne_anonymize`?

It's easy enough to add it to mne-python in any case. It could be an argument to `.anonymize(meas_date=None)` with a deprecation cycle False->True behavior. @agramfort do you think we should add it?

agramfort, 2016-03-07:
mne-c does not do it but that seems legit

@mshamalainen thoughts?

mshamalainen, 2016-03-19:
When I wrote me_anonymize, I didn't think the measurement date would be PHI.

agramfort, 2016-03-19:
Ok. Can you take care of amending the mne anonymize function and we ll do the same in the python code?

mshamalainen, 2016-03-21:
me_anonymize now removes FIFF_MEAS_DATE as well

Eric89GXL, 2016-03-21:
@mshamalainen shoudl we also remove the SECS, MSECS, and USECS from FILE_ID and MEAS_ID to remove traces of the recording date? That's what I did in  #3041